orangepi-xunlong/orangepi-build #2: Orange pi zero 2 can't reboot

chichivica, 2021-03-01:
Hello,
I recently bought orangepi zero 2 sample and there is most strage thing: it doesn't reboot on 
```
sudo reboot now
```
or 
```
sudo shutdown -r now
```
it actually shuts down and never powers up.

This happens with these firmware:
- Orangepizero2_2.1.0_ubuntu_focal_server_linux4.9.170
- Armbian_21.02.1_Orangepizero2_focal_current_5.10.12
- Armbian_21.02.1_Orangepizero2_buster_current_5.10.12

But with these firmware reboot works properly:
- Orangepizero2_2.1.0_ubuntu_bionic_server_linux4.9.170
- Orangepizero2_2.1.0_debian_buster_server_linux4.9.170

I tested this with many power supplies.

orangepi-xunlong, 2021-03-04:
It's strange that everything is normal here(use **Orangepizero2_2.1.0_ubuntu_focal_server_linux4.9.170**). You may need to find other reasons, thanks!

[image]

kasinjsh, 2021-03-05:
I have this issue too. H6 have broken Watchdog implementation on some units, so looks like H616 have that too - we are "the lucky ones".

http://linux-sunxi.org/H6 (refference)

orangepi-xunlong, 2021-03-08:
Hi, 
     At present, only a certain firmware(Ubuntu Focal) has reported a problem (and my test is still fine, and no one else has reported any problem with reboot), so we can't conclude that there is a problem with the watchdog and more information is needed (besides, my H6 reboot test is also fine, It would be nice if there were steps to repeat the reboot problem).

kasinjsh, 2021-03-08:
> Hi,
> At present, only a certain firmware(Ubuntu Focal) has reported a problem (and my test is still fine, and no one else has reported any problem with reboot), so we can't conclude that there is a [...]

Here is video of what is happening: https://cloud.harijs.id.lv/index.php/s/5QGMs2Ys5RDNk99 . I have tried all OS version, updated bootloader, tried different chargers.

Logs from /var/log: https://cloud.harijs.id.lv/index.php/s/gPp7CDErDe2Qptx

orangepi-xunlong, 2021-03-09:
This is my test situation, after reboot can boot normally.

orangepi-xunlong, 2021-03-09:
Hi, 

Thank you for your feedback. This is my test video, after reboot can boot normally: [orangepizero2_reboot.mp4](https://drive.google.com/file/d/147ru1M7Bm1bNyQ6_WFmLG0--T3rIwaS9/view?usp=sharing).

I have seen the video you sent, and there should be something wrong in the U-Boot stage after the reboot. If you can connect the debugging serial port, you should be able to clearly locate the cause of the error.

The only difference I can think of now is that the SD cards we use may be different (we know that there are problems with non SanDisk cards)

Can you provide the model of SD card you are using, or can you test it with SanDisk SD card  shown in the figure below, thanks!
[image]

closed 2021-03-12 by orangepi-xunlong

Darkyere, 2022-07-01:
Hi i have installed Armbian Jammy 22.05 on my Orange Pi Zero 2 yesterday.
I am also experiencing reboot issues. It shuts down, turn off lights but never boots.
I use an sdcard that looks exactly like the image u send

[image]

I have a serial connection, so i will be able to do some debugging of the kind u should need.

zorflieg, 2022-09-16:
I'm running Armbian bullseye on zero 2 and cannot reboot either.  Did this get looked at?

Darkyere, 2022-09-16:
Still doesn't work for me.

mallesh0403, 2022-09-16:
Hi, I'm using openwrt on Orange Pi zero. when I've run the reboot -f command I'm also facing the same issue. the Orangepi zero was not booted successfully.

javilondo, 2022-09-26:
I have the same issue too with both images:
Armbian_22.08.1_Orangepizero2_bullseye_current_5.15.63.img
Armbian_22.08.1_Orangepizero2_jammy_current_5.15.63.img

aminismail, 2023-03-25:
Has anyone resolved this?

mikerodrigues, 2023-07-26:
Also seeing this issue. Have not hooked up to serial or verified SD card yet.

qrsforever, 2023-08-11:
same issue:
sandisk 64GB
Orangepizero2_3.0.6_ubuntu_jammy_server_linux5.16.17

here is dmesg log:
```
[   13.204615] unisoc_wifi unisoc_wifi wlan0: mixed HW and IP checksum settings.
[   13.276844] Unable to handle kernel paging request at virtual address ffff80000925ec00
[   13.276867] Mem abort info:
[   13.276870]   ESR = 0x96000007
[   13.276875]   EC = 0x25: DABT (current EL), IL = 32 bits
[   13.276882]   SET = 0, FnV = 0
[   13.276887]   EA = 0, S1PTW = 0
[   13.276892]   FSC = 0x07: level 3 translation fault
[   13.276897] Data abort info:
[   13.276900]   ISV = 0, ISS = 0x00000007
[   13.276904]   CM = 0, WnR = 0
[   13.276909] swapper pgtable: 4k pages, 48-bit VAs, pgdp=00000000413bd000
[   13.276917] [ffff80000925ec00] pgd=100000007ffff003, p4d=100000007ffff003, pud=100000007fffe003, pmd=100000007fffa003, pte=0000000000000000
[   13.276943] Internal error: Oops: 96000007 [#1] SMP
[   13.281842] Modules linked in: cpufreq_dt(+) zram sprdwl_ng nfsd sunxi_addr auth_rpcgss cfg80211 nfs_acl lockd grace mtdpstore sunrpc ramoops pstore_blk fuse reed_solomon pstore_zone realtek i2c_mv64xxx dwmac_sun8i mdio_mux
[   13.301728] CPU: 2 PID: 357 Comm: systemd-udevd Not tainted 5.16.17-sun50iw9 #3.0.6
[   13.309396] Hardware name: OrangePi Zero2 (DT)
[   13.313847] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   13.320814] pc : cpufreq_generic_get+0x24/0xa8
[   13.325280] lr : cpufreq_online+0x410/0x9d0
[   13.329474] sp : ffff80000a4737a0
[   13.332792] x29: ffff80000a4737a0 x28: ffff000004d52408 x27: 0000000000000001
[   13.339939] x26: ffff000004d525d0 x25: ffff8000094d8790 x24: ffff80000957c000
[   13.347089] x23: ffff80000943bd48 x22: ffff000004d52490 x21: 0000000000000000
[   13.354234] x20: 0000000000000000 x19: ffff000004d52400 x18: 0000000000000000
[   13.361380] x17: 2e726f74616c7567 x16: 65723a726f74616c x15: ffff000004e19fc0
[   13.368527] x14: ffffffffffffffff x13: ffff000002990588 x12: ffff000002990586
[   13.375676] x11: 0000000000000030 x10: 0101010101010101 x9 : 0000000000000000
[   13.382822] x8 : ffff000002990748 x7 : ffff000002990758 x6 : ffff000004d524e8
[   13.389968] x5 : 000000000000000f x4 : 000000000000000f x3 : 000000000000000f
[   13.397116] x2 : ffff800000f80070 x1 : ffff80000925e7c0 x0 : 0000000000000440
[   13.404264] Call trace:
[   13.406721]  cpufreq_generic_get+0x24/0xa8
[   13.410835]  cpufreq_online+0x410/0x9d0
[   13.414683]  cpufreq_add_dev+0xbc/0xd0
[   13.418444]  subsys_interface_register+0x9c/0xf8
[   13.423077]  cpufreq_register_driver+0x1b0/0x388
[   13.427708]  dt_cpufreq_probe+0x200/0x448 [cpufreq_dt]
[   13.432869]  platform_probe+0x68/0xd8
[   13.436544]  really_probe+0xbc/0x428
[   13.440133]  __driver_probe_device+0x114/0x188
[   13.444588]  driver_probe_device+0x80/0x110
[   13.448784]  __driver_attach+0xbc/0x1a0
[   13.452632]  bus_for_each_dev+0x70/0xc0
[   13.456482]  driver_attach+0x24/0x30
[   13.460069]  bus_add_driver+0x140/0x220
[   13.463916]  driver_register+0x64/0x120
[   13.464233] systemd[1]: Found device /dev/ttyS0.
[   13.467764]  __platform_driver_register+0x28/0x38
[   13.467777]  dt_cpufreq_platdrv_init+0x24/0x1000 [cpufreq_dt]
[   13.467792]  do_one_initcall+0x54/0x1b8
[   13.467803]  do_init_module+0x44/0x1f0
[   13.467812]  load_module+0x20e0/0x2850
[   13.494223]  __do_sys_finit_module+0xbc/0x128
[   13.494248]  __arm64_sys_finit_module+0x20/0x30
[   13.494256]  invoke_syscall+0x44/0x108
[   13.506914]  el0_svc_common.constprop.0+0xcc/0xf0
[   13.511652]  do_el0_svc+0x24/0x88
[   13.511670]  el0_svc+0x20/0x60
[   13.511679]  el0t_64_sync_handler+0xb0/0xb8
[   13.511686]  el0t_64_sync+0x1a0/0x1a4
[   13.511700] Code: f0003f81 911f0021 910003fd f8635800 (f8616800) 
[   13.532025] ---[ end trace 0d8a5d59cabce41a ]---
[   13.664674] mtty_probe init device addr: 0x000000003d3faf4a
[   13.667882] systemd[1]: Reached target Hardware activated USB gadget.
[   13.670614] -->rfkill_bluetooth_init
[   13.680615] bluetooth_set_power: start_block=1
[   13.680631] WCN: marlin power state:4, subsys: [MARLIN_BLUETOOTH] power 0
[   13.691930] WCN: can not power off, other module is on
[   13.691943] bluetooth_set_power: end_block=1
[   13.701426] <--rfkill_bluetooth_init
[   13.709954] systemd[1]: Finished Load Kernel Modules.
[   13.720997] systemd[1]: Starting Apply Kernel Variables...
[   14.096321] debugfs: Directory '5096000.codec' with parent 'H616 Audio Codec' already present!
[   14.098470] zram0: detected capacity change from 0 to 1007736

```

aidestob, 2023-08-26:
Hi. Maybe this will help you https://giters.com/orangepi-xunlong/orangepi-build/issues/50

aminismail, 2023-08-26:
> Hi. Maybe this will help you https://giters.com/orangepi-xunlong/orangepi-build/issues/50

Thanks for the tip. However, I am using a OP Zero LTS and the file does not apply to it.
 
Actually, I worked around the issue by downgrading to Xenial. The problem seems to occur with Bionic (at least on the OP Zero LTS) but does not on the Xenial image. I have verified this several times.

For now, the differences between the two OSs don't matter to me, so I'm good.

Thank you.