orangepi-xunlong/orangepi-build #4: Change U-boot args

AndreiCherniaev, 2021-04-13:
Hello

I rebuild U-boot for my Orange pi 4B. I want disable debug uart. As I see in file boot-rockchip64.cmd I shouldn't change anything. I need /boot/orangepiEnv.txt But I can't find it in my build tree. @orangepi-xunlong are you shure that  /boot/orangepiEnv.txt exist for Orange Pi 4B?

[image]

orangepi-xunlong, 2021-04-13:
Each platform has a slightly different version of orangepiEnv.txt. Here is the RK3399:

https://github.com/orangepi-xunlong/orangepi-build/blob/main/external/config/bootenv/rockchip.txt

It will be copied at compile time as /boot/orangepiEnv.txt

closed 2021-04-13 by orangepi-xunlong

AndreiCherniaev, 2021-04-20:
> Each platform has a slightly different version of orangepiEnv.txt. Here is the RK3399:
> 
> https://github.com/orangepi-xunlong/orangepi-build/blob/main/external/config/bootenv/rockchip.txt
> 
> It will be copied at compile time as /boot/orangepiEnv.txt

How I was understand that orangepi-build/external/config/bootenv/rockchip.txt is orangepiEnv.txt for my Orange Pi 4B? How I can guess name for another SBC like Orange Pi 4g or Orange Pi PC PLUS?

AndreiCherniaev, 2021-04-20:
I try new Ubuntu18 img from googledrive for Orange Pi 4B. There is no ttyS4, but there is ttyS0 and ttyFIQ0. But as I understand tty0 (means UART0) is part of bluetooth module, isn't in @orangepi-xunlong ?
I also try build Ubuntu 18 from [orangepi-build](https://github.com/orangepi-xunlong/orangepi-build). It has got the same problem.

orangepi-xunlong, 2021-04-21:
> 
> 
> I try new Ubuntu18 img from googledrive for Orange Pi 4B. There is no more ttyS4, but there is ttyS0 and ttyFIQ0. But as I understand tty0 (means UART0) is part of bluetooth module, isn't in @orangepi-xunlong ?

Yes, you are right!

> I also try build Ubuntu 18 from [orangepi-build](https://github.com/orangepi-xunlong/orangepi-build). It has got the same problem. UART0 is not in 40 pins GPIO so I can't use it... I also look at OrangePi RK3399... In OrangePi RK3399 there is UART2 for debug and there is UART0 on 40 pins GPIO. So looks like @orangepi-xunlong using dts file of OrangePi RK3399 for Orange Pi 4B.
> 
> P.S. In old Ubuntu18 there is ttyS4.
> P.S.S. My dream is to disable debug UART (ttyFIQ0) to use it in my application. How I can do it?

UART4 and SPI1 are multiplexed. By default, SPI1 is turned on and UART4 is turned off. You can turn off SPI1 and turn on UART4 in [DTS](https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-4.4-rockchip64/arch/arm64/boot/dts/rockchip/rk3399-orangepi-common.dtsi), so that UART4 in 40PIN can be used.

[image]

This is the procedure in the Chinese manual. I believe you can understand it. Good luck!

AndreiCherniaev, 2021-04-21:
> Yes, you are right!
>
>
> UART4 and SPI1 are multiplexed. By default, SPI1 is turned on and UART4 is turned off. You can turn off SPI1 and turn on UART4 in [...]

There is **no** ... dts/rockchiprk3399-orangepi-common.dts

orangepi-xunlong, 2021-04-22:
> 
> 
> > > I try new Ubuntu18 img from googledrive for Orange Pi 4B. There is no more ttyS4, but there is ttyS0 and ttyFIQ0. But as I understand tty0 (means UART0) is part of bluetooth module, isn't in @orangepi-xunlong ?
> > 
> > 
> > Yes, you are right!
> > > I also try build Ubuntu 18 from [orangepi-build](https://github.com/orangepi-xunlong/orangepi-build). It has got the same problem. UART0 is not in 40 pins GPIO so I can't use it... I also look at OrangePi RK3399... In OrangePi RK3399 there is UART2 for debug and there is UART0 on 40 pins GPIO. So looks like @orangepi-xunlong using dts file of OrangePi RK3399 for Orange Pi 4B.
> > > P.S. In old Ubuntu18 there is ttyS4.
> > > P.S.S. My dream is to disable debug UART (ttyFIQ0) to use it in my application. How I can do it?
> > 
> > 
> > UART4 and SPI1 are multiplexed. By default, SPI1 is turned on and UART4 is turned off. You can turn off SPI1 and turn on UART4 in [DTS](https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-4.4-rockchip64/arch/arm64/boot/dts/rockchip/rk3399-orangepi-common.dtsi), so that UART4 in 40PIN can be used.
> > [image]
> > This is the procedure in the Chinese manual. I believe you can understand it. Good luck!
> 
>     1. There is **no** ... dts/rockchiprk3399-orangepi-common.dts
> 
> 
> May be you are talking about rk3399-orangepi-common.dtsi? There are two rk3399-orangepi-common.dtsi:
> in boot/dts/rockchip
> and in /scripts/dts/include-prefix/arm64/rockchip
> what item I need?

Sorroy, it should be rk3399-orangepi-common.dtsi. 

>     1. Why I can't edit u-boot/v2020.04-rockchip64/u-boot.cfg? After every make u-boot (compilation.sh) file u-boot.cfg will regenerating, all my changes will erased! For example I want do
>        #define CONFIG_CMD_CONSOLE **0**
>        How?

U-boot.cfg is automatically generated and you should modify configs/orangepi-4-rk3399_defconfig, for example:

```

--- a/configs/orangepi-4-rk3399_defconfig
+++ b/configs/orangepi-4-rk3399_defconfig
@@ -11,6 +11,7 @@ CONFIG_DEBUG_UART=y
 CONFIG_DEFAULT_FDT_FILE="rockchip/rk3399-orangepi-4.dtb"
 CONFIG_MISC_INIT_R=y
 # CONFIG_DISPLAY_CPUINFO is not set
+# CONFIG_CMD_CONSOLE is not set
 CONFIG_DISPLAY_BOARDINFO_LATE=y
 # CONFIG_SPL_RAW_IMAGE_SUPPORT is not set
 CONFIG_SPL_STACK_R=y

```