orangepi-xunlong/orangepi-build #5: Burn new U-BOOT on Orange Pi 4B

AndreiCherniaev, 2021-04-14:
I build new U-BOOT for my Orange Pi 4B. There are a lot of files at my screenshort. How I should transfer my new U-BOOT to my 32GB microSD? What I should write in dd command?
[image]
[image]

AndreiCherniaev, 2021-04-16:
I try build Ubuntu 18 for Orange Pi PC Plus. I buils full OS and 
sudo dd
and for Orange Pi PC Plus works ok. But for Orange Pi 4B looks like there are **no** u-boot on .img.

AndreiCherniaev, 2021-04-19:
@longerGG @leeboby @orangepi-xunlong @SpadeLushen any ideas?

kasinjsh, 2021-04-19:
I have this issue on Zero2 also, no lights light up on startup, so it uboot. I tried compiling multiple distros, all have the same issue, v2.1.2.

orangepi-xunlong, 2021-04-20:
> 
> 
> I build new U-BOOT for my Orange Pi 4B. There are a lot of files at my screenshort at figure 3. How I should transfer my new U-BOOT to my 32GB microSD? What I should write in dd command?
> 
> At this time I use official Ubuntu 18 from xunlong google drive on my microsd look at figure 1.
> [image]
> Figure 1. xunlong Ubuntu 18 on microsd
> [image]

Figure 1 should be the old image. The new image compiled by orangepi-build has been uploaded. Please test with the new image, this way, the u-boot that orangepi-build compiled can be directly updated
[image]

> Figure 2. My Ubuntu 18 on microsd **(where u-boot?)**
> [image]
> Figure 3. Building Ubuntu process
> 
> I also try "Full OS image for flashing" but in folder output/images there is Orangepi4_2.1.2_ubuntu_bionic_server_linux4.4.179_minimal.img only, **no** Orangepi4_2.1.2_ubuntu_bionic_server_linux4.4.179_minimal.**tar.gz**. I try flash to my 32GB microsd
> sudo dd if=Orangepi4_2.1.2_ubuntu_bionic_server_linux4.4.179_minimal.img of=/dev/sdb
> 
> **but no boot sectors in Orangepi4_2.1.2_ubuntu_bionic_server_linux4.4.179_minimal.img (only one ext4 sector about 2GB) and Orange Pi 4B do nothing.**

1. I have tested the image of OrangPi4 compiled by orangepi-build, and it can boot normally. Please use etcher to re-burn and test it.
2. Please refer to the [manual of orangepizero2](https://drive.google.com/drive/folders/1ToDjWZQptABxfiRwaeYW1WzQILM5iwpb) for how to update u-boot(Compile u-boot chapter), OrangePi4's manual is still being written and will be released soon.

[image]

orangepi-xunlong, 2021-04-20:
> I have this issue on Zero2 also, no lights light up on startup, so it uboot. I tried compiling multiple distros, all have the same issue, v2.1.2.

Please update the orangepi-build code to its latest state.

[b4998ad13d776264b561c7e06ece2b9734e7d436](https://github.com/orangepi-xunlong/orangepi-build/commit/b4998ad13d776264b561c7e06ece2b9734e7d436)  

Fixed an issue where Orangepi Zero 2 U-boot could not boot.

AndreiCherniaev, 2021-04-20:
My way to burn Ubuntu img with dd has a mistake. I don't know why 
`sudo dd if=Orangepi4_2.1.2_ubuntu_bionic_server_linux4.4.179_minimal.img of=/dev/sdb`
is incorrect, but now I use [balena](https://www.balena.io/etcher/). Using balena I can burn Ubuntu img on my microsd card and it will really works! Thanks for help @orangepi-xunlong !

closed 2021-04-20 by AndreiCherniaev