orangepi-xunlong/orangepi-build #8: apt-get upgrade

AndreiCherniaev, 2021-04-23:
After build full OS using [orangepi-build](https://github.com/orangepi-xunlong/orangepi-build/issues/new) I try upgrade packages of my Ubuntu 18
The end of log is
```
The following packages will be upgraded:
  libcaca0
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 192 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://mirrors.tuna.tsinghua.edu.cn/ubuntu-ports bionic-security/main arm64 libcaca0 arm64 0.99.beta19-2ubuntu0.18.04.2 [192 kB]
Fetched 192 kB in 3s (76.8 kB/s)   
(Reading database ... 33743 files and directories currently installed.)
Preparing to unpack .../libcaca0_0.99.beta19-2ubuntu0.18.04.2_arm64.deb ...
Unpacking libcaca0:arm64 (0.99.beta19-2ubuntu0.18.04.2) over (0.99.beta19-2ubuntu0.18.04.1) ...
Setting up libcaca0:arm64 (0.99.beta19-2ubuntu0.18.04.2) ...
Processing triggers for libc-bin (2.27-3ubuntu1.4) ...
/sbin/ldconfig.real: /usr/lib/libdrm.so.2 is not a symbolic link

/sbin/ldconfig.real: /usr/lib/liborc-test-0.4.so.0 is not a symbolic link

/sbin/ldconfig.real: /usr/lib/libdrm_radeon.so.1 is not a symbolic link

/sbin/ldconfig.real: /usr/lib/libdrm_amdgpu.so.1 is not a symbolic link

/sbin/ldconfig.real: /usr/lib/libdrm_freedreno.so.1 is not a symbolic link

/sbin/ldconfig.real: /usr/lib/libdrm_nouveau.so.2 is not a symbolic link

/sbin/ldconfig.real: /usr/lib/libkms.so.1 is not a symbolic link

/sbin/ldconfig.real: /usr/lib/liborc-0.4.so.0 is not a symbolic link

```

I try build Qt with KMS. This is a big problem because "/sbin/ldconfig.real: /usr/lib/libkms.so.1 is not a symbolic link"
How I can solve it?

orangepi-xunlong, 2021-04-23:
Sorry, this goes beyond the problem orangepi-build needs to solve, and Google may be a good solution.

closed 2021-04-23 by orangepi-xunlong

AndreiCherniaev, 2021-04-27:
@orangepi-xunlong there wasn't same problem on previous Ubuntu 18 build! You can verify that no problem using backup image from your [googledrive](http://www.orangepi.org/downloadresources/OrangePi4B/2020-01-03/orangepi3_f9433ee3d9493fd6fc711c065d.html). 

I am googling a lot of but I can't solve problems. Please help.

orangepi-xunlong, 2021-04-28:
How to repeat, please send the detailed steps, thanks!

AndreiCherniaev, 2021-05-04:
> How to repeat, please send the detailed steps, thanks!

1) Build your Ubuntu 18 with graphics
2) open console and 
`sudo /sbin/ldconfig`
so you can see problems... 
3) Try to do this steps also on your old Ubuntu 18 build!

orangepi-xunlong, 2021-05-06:
The issue has been resolved and you can download the latest image for testing.

If you compile the image yourself, go to the following directory and update the code with the **git pull** command, then recompile the linux image.

`orangepi-build/external/cache/sources/rk3399_gst_xserver_libs`