orangepi-xunlong/orangepi-build #12: How to enable mpp hardware encode/decode function in the RK3328(ubuntu 20.04)(kernel branch:"orange-pi-5.8-rockchip64")

zhangweiwahaha042, 2021-08-16:
■Summary
・hardware platform
Orange Pi R1 Plus(RK3328)
・kernel source and branch
https://github.com/orangepi-xunlong/linux-orangepi
※branch:orange-pi-5.8-rockchip64
・mpp library source
https://github.com/rockchip-linux/mpp/
・issue
the error occur because the following 3 devices can not be found when run the mpp library
①/dev/vpu_service
②/dev/dri/card0
③/dev/ion
・additional information
the kernel version of the Android9 of the RK3328 is the 4.4 and the following 2 devices can be found
①/dev/vpu_service
②/dev/dri/card0

■Question
Is the following good solution what enable mpp hardware encode/decode function in the RK3328(ubuntu 20.04)(kernel branch:"orange-pi-5.8-rockchip64")?
transplant the driver vpu_service and the driver dri from the kernel 4.4 of the Android9 to the kernel 5.8 of ubuntu 20.04.
※If you have the related patchs, could you release it to us?

Or, could you give us some good suggestions?

Thank you very much.

zhangweiwahaha042, 2021-08-16:
In order to good communication, I also write the above QA again by the chinese here.

■概要
・硬件平台
Orange Pi R1 Plus(RK3328)
・内核代码和分支
https://github.com/orangepi-xunlong/linux-orangepi
※分支名：orange-pi-5.8-rockchip64
・mpp库源代码
https://github.com/rockchip-linux/mpp/
・错误现象
在上述硬件平台上，我们想使用mpp的硬解码功能，但是在运行mpp库时，报了错误，显示下面3个设备节点没有生成
①/dev/vpu_service
②/dev/dri/card0
③/dev/ion
・补充信息
上述硬件平台的Android9的linux内核版本是4.4，有下面2个设备节点
①/dev/vpu_service
②/dev/dri/card0

■问题
为了在RK3328的ubuntu20.04上支持mpp解码功能，我们应该怎么做比较好呢？我们现在想到的方向如下，你们觉得是否可行？
・方案①
将Android9上内核版本4.4的vpu_service驱动、dri驱动移植到ubuntu20.04上内核版本5.8上
※如果贵公司有相关补丁提供的话，不胜感激
・方案②
贵公司有没有其他的推荐方案

orangepi-xunlong, 2021-08-16:
1. linux-5.8据我所知，是无法使用RK官方的mmp库，我们没有在R1Plus上调试过这个功能，也不打算支持这个功能，所以没有补丁提供
2. 如果需要使用，可以考虑自己移植下linux4.4的内核（由于R1Plus功能很少，移植应该不难），另外github上应该有一些适配了RK3328的linux4.4的代码，可以直接参考下，谢谢

closed 2021-08-16 by orangepi-xunlong

zhangweiwahaha042, 2021-08-16:
非常感谢您的回复，我们这边试验下移植linux4.4的方案。