orangepi-xunlong/orangepi-build #18: R1 PLUS LTS 编译错误，按照OrangePi_R1_Plus_LTS_RK3328_用户手册_v1.4.pdf这个文档的方法，从百度云下载的openwrt，编译出错，下面是错误log：

ShareCat, 2021-12-06:
Scanning dependencies of target rpcd-mod-luci
make[6]: Leaving directory '/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614'
make[6]: Entering directory '/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614'
[ 50%] Building C object CMakeFiles/rpcd-mod-luci.dir/luci.c.o
/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614/luci.c: In function 'rpc_luci_get_iwinfo':
/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614/luci.c:938:14: error: 'IWINFO_80211_AX' undeclared (first use in this function); did you mean 'IWINFO_80211_A'?
   if (nret & IWINFO_80211_AX)
              ^~~~~~~~~~~~~~~
              IWINFO_80211_A
/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614/luci.c:938:14: note: each undeclared identifier is reported only once for each function it appears in
/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614/luci.c:983:14: error: 'IWINFO_HTMODE_HE20' undeclared (first use in this function); did you mean 'IWINFO_HTMODE_HT20'?
   if (nret & IWINFO_HTMODE_HE20)
              ^~~~~~~~~~~~~~~~~~
              IWINFO_HTMODE_HT20
/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614/luci.c:986:14: error: 'IWINFO_HTMODE_HE40' undeclared (first use in this function); did you mean 'IWINFO_HTMODE_HT40'?
   if (nret & IWINFO_HTMODE_HE40)
              ^~~~~~~~~~~~~~~~~~
              IWINFO_HTMODE_HT40
/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614/luci.c:989:14: error: 'IWINFO_HTMODE_HE80' undeclared (first use in this function); did you mean 'IWINFO_HTMODE_VHT80'?
   if (nret & IWINFO_HTMODE_HE80)
              ^~~~~~~~~~~~~~~~~~
              IWINFO_HTMODE_VHT80
/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614/luci.c:992:14: error: 'IWINFO_HTMODE_HE160' undeclared (first use in this function); did you mean 'IWINFO_HTMODE_VHT160'?
   if (nret & IWINFO_HTMODE_HE160)
              ^~~~~~~~~~~~~~~~~~~
              IWINFO_HTMODE_VHT160
CMakeFiles/rpcd-mod-luci.dir/build.make:81: recipe for target 'CMakeFiles/rpcd-mod-luci.dir/luci.c.o' failed
make[6]: *** [CMakeFiles/rpcd-mod-luci.dir/luci.c.o] Error 1
make[6]: Leaving directory '/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614'
CMakeFiles/Makefile2:94: recipe for target 'CMakeFiles/rpcd-mod-luci.dir/all' failed
make[5]: *** [CMakeFiles/rpcd-mod-luci.dir/all] Error 2
make[5]: Leaving directory '/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614'
Makefile:148: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614'
Makefile:54: recipe for target '/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614/.built' failed
make[3]: *** [/home/user/openwrt/openwrt/build_dir/target-aarch64_generic_musl/rpcd-mod-luci-20210614/.built] Error 2
make[3]: Leaving directory '/home/user/openwrt/openwrt/feeds/luci/libs/rpcd-mod-luci'
time: package/feeds/luci/rpcd-mod-luci/compile#0.18#0.09#0.27
    ERROR: package/feeds/luci/rpcd-mod-luci failed to build.
package/Makefile:112: recipe for target 'package/feeds/luci/rpcd-mod-luci/compile' failed
make[2]: *** [package/feeds/luci/rpcd-mod-luci/compile] Error 1
make[2]: Leaving directory '/home/user/openwrt/openwrt'
package/Makefile:108: recipe for target '/home/user/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/user/openwrt/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/openwrt/openwrt'
/home/user/openwrt/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

ShareCat, 2021-12-06:
编译前需要执行：
```shell
cp configs/OrangePi_R1_Plus_LTS_defconfig .config
```
然而configs路径下没有 OrangePi_R1_Plus_LTS_defconfig .config 这个文件，只有 OrangePi_R1Plus_defconfig 这个文件

ShareCat, 2021-12-06:
该更新的也更新了

```shell
user@user-virtual-machine:~/openwrt/openwrt$ ./scripts/feeds update -a
Updating feed 'packages' from 'https://git.openwrt.org/feed/packages.git;openwrt-21.02' ...
Already up to date.
Create index file './feeds/packages.index' 
Updating feed 'luci' from 'https://git.openwrt.org/project/luci.git;openwrt-21.02' ...
Already up to date.
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git;openwrt-21.02' ...
Already up to date.
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git;openwrt-21.02' ...
Already up to date.
Create index file './feeds/telephony.index' 
user@user-virtual-machine:~/openwrt/openwrt$ ./scripts/feeds install -a
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
```

orangepi-xunlong, 2021-12-06:
1. 百度云的代码还不是最新的，编译前先使用git pull把代码更新下再按照手册操作，也可以直接去 
[ https://github.com/orangepi-xunlong/openwrt](https://github.com/orangepi-xunlong/openwrt)
这里下载最新的代码
2. 如果后面还有问题请把issue提到 
[ https://github.com/orangepi-xunlong/openwrt](https://github.com/orangepi-xunlong/openwrt)
这个仓库

谢谢！

closed 2021-12-06 by orangepi-xunlong