orangepi-xunlong/orangepi-build #20: Orangepizero2_2.1.6_ubuntu_focal_server_linux5.13.0 ble issue

Shandawei, 2021-12-20:
with `Orangepizero2_2.1.6_ubuntu_focal_server_linux5.13.0`
with `bluepy`
```python
from bluepy.btle import DefaultDelegate, Peripheral

try:
    d= Peripheral("80:6f:b0:b7:72:b0", "public")
    print(d)
except Exception as e:
    print(e)
```
cannot connect the ble e device ,the code above allways raise Exception

with `Orangepizero2_2.1.6_ubuntu_focal_server_linux4.9.170`
Everything goes well.

orangepi-xunlong, 2021-12-23:
Hi,

Linux5.13 ble is also unavailable in our test. Please use linux4.9 first, thanks!

Shandawei, 2021-12-23:
> Hi,
> 
> Linux5.13 ble is also unavailable in our test. Please use linux4.9 first, thanks!

Thanks,I get it.

closed 2022-04-08 by orangepi-xunlong

XNicON, 2022-06-24:
BLE me work on Orangepizero2_3.0.4_ubuntu_jammy_server_linux5.16.17