orangepi-xunlong/orangepi-build #22: Uboot compilation failed

afaulkner420, 2022-01-09:
[image]

I'm getting this on build. Any ideas?

afaulkner420, 2022-01-09:
It seems this error occurs only when building the legacy kernel for ubuntu focal, for orange pi 3 lts, with desktop environment. 
If I build the mainline kernel, then it passes this error, and compiles. 
Not sure if this helps with the error at all?

afaulkner420, 2022-01-09:
I've uploaded the set of logs from output/debug incase anyone wants to have a look and try to help?

[1.zip](https://github.com/orangepi-xunlong/orangepi-build/files/7835255/1.zip)

orangepi-xunlong, 2022-01-10:
Hello, can you test orangepi-build on Ubuntu18.04, Ubuntu20.04 is not tested. Thanks!

[image]

afaulkner420, 2022-01-10:
Ah brilliant! Yes I shall try this and get back to you! 
Is there a command to clean the build environment? Or is a separate build folder recommended for every different build?

afaulkner420, 2022-01-10:
Just waiting on purchasing a board and then I'll be testing my builds. I want to maintain this hoard if possible. There's not a lot of info about it on the Internet. I want to be the first. Maybe port to armbian aswell! 
Thanks for all help! Very much appreciated!

afaulkner420, 2022-01-10:
Well that seems to have worked perfectly! Tha k you 
[image]

By the way, which has better support for the 3-lts, mainline or legacy kernel? Does everything work as it should for both variants?

orangepi-xunlong, 2022-01-11:
Legacy kernel is probably better from a driver standpoint, linux5.10 does not have Audio Codec drivers. 

For more information, you can read the user manual above, thanks!

closed 2022-01-11 by orangepi-xunlong

afaulkner420, 2022-01-11:
The audio codec drivers. Is this due to 5.10 breaking them with the regmap_write issue?
As I believe this is fixed now? Or is there other issues?

orangepi-xunlong, 2022-01-11:
It has nothing to do with regmap, there is no driver for allwinner h6 audio codec.

afaulkner420, 2022-01-11:
Ironically I just found this out. 

There does seem to be work going on for it though. 
https://www.spinics.net/lists/arm-kernel/msg819709.html

afaulkner420, 2022-01-11:
Also, just noticed, does the hdmi-audio work on h6?

https://github.com/PopcornComputer/PocketPC-Linux/commit/4e3ad68c2d8fb9cb69af8bcedd49ef76dfbae200

Looking into this commit, you have half the commit I your 5.10 kernel. I think with a few alterations and patches, it could work. Then just audio jack to fix. Guessing this is the DAC side that isn't working. Again, I believe work is being done again for this

orangepi-xunlong, 2022-01-12:
Yes, HDMI audio is fine in linux5.10.