orangepi-xunlong/orangepi-build #24: Bluetooth doesn't work on OPI3 LTS

stijn-devriendt, 2022-02-14:
Hi,

The image built by current mainline doesn't include hciattach_opi:
During build it shows:
```
install: cannot stat '/home/sde/projects/orangepi-build/external/packages/blobs/bt/hciattach/hciattach_opi_arm64_current': No such file or directory
```

It looks like this patch (ef22bd4b078388eab200a68384a6dff8d959dcb0) did not include the same change for the OPI3 hardware:
https://github.com/orangepi-xunlong/orangepi-build/blob/main/external/config/sources/families/sun50iw6.conf

Regards,
Stijn

orangepi-xunlong, 2022-02-15:
Hello, 

It's fixed, you can update the orangepi-build and test it again. Thank you!

stijn-devriendt, 2022-02-15:
Thank you for your quick turnaround. I've rebuilt the RFS and the mentioned issues are fixed.

Unfortunately, there remains a problem that bluetooth low-energy doesn't appear to function properly. When a scan is started, no BLE devices are found:
```
>sudo apt-get install -u bluez
> bluetoothctl
# menu scan
# transport le
# back
# scan on
```
does not detect BLE devices.
However:
```
hcitool lescan
```
does detect them.

Evidence points towards a bug in the driver, since bluez uses the MGMT protocol to start a scan while hcitool uses raw HCI commands. It looks like the translation from mgmt protocol to hci in the kernel (driver) is broken for LE scanning.