orangepi-xunlong/orangepi-build #29: OPi Zero 2 GPU Drivers?

NetuserPRO46, 2022-04-18:
Hi, i have a problem with mainline kernel panfrost driver. I'm compiled it into kernel and `dmesg | grep panfrost` shows nothing.
However, official arm mali driver doesn't work too. System just can't detect gpu. I think there's problem in device tree.

orangepi-xunlong, 2022-04-19:
Hi, which code are you using?

orangepi-build next branch: Debian12 Linux5.16 GPU is ok.

[image]

[image]

[image]

[image]

[image]

NetuserPRO46, 2022-04-19:
How do i compile the 5.16 kernel?

NetuserPRO46, 2022-04-19:
Thanks for reply!

closed 2022-04-19 by NetuserPRO46