orangepi-xunlong/orangepi-build #33: No have sound in HDMI Orange Pi Zero 2 Debian XFCE 3.0.0

agevlakh, 2022-05-17:
Hello! I install on orangepi zero 2 this image: Orangepizero2_3.0.0_debian_bullseye_desktop_xfce_linux5.16.17
In this image I havent sound in HDMI. Dmesg show next errors:
input: dmesg | grep snd
[    1.291730] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.235495] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.236540] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.243243] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.243401]   #0: sndahub
[    2.703637] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.705602] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed
[    2.729792] sunxi-hdmi soc:hdmi-audio@1: snd_soc_register_card failed

pulseaudio -v  command have this error:
E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="0" name="platform-soc_sound_0" card_name="alsa_card.platform-soc_sound_0" namereg_fail=false tsched=no fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.
I: [pulseaudio] module-udev-detect.c: Card /devices/platform/soc/soc:sound@0/sound/card0 (alsa_card.platform-soc_sound_0) failed to load module.

Please fix this error.
Thanks!

orangepi-xunlong, 2022-05-18:
`aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav`

Does this have a sound? 

Thanks!

agevlakh, 2022-05-18:
> `aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav`
> 
> Does this have a sound?
> 
> Thanks!

I test the command. No have sound. But in terminal I havent errors. It was clear that music was being played, but no sound was being output. My hdmi cable is working. For example in Orangepizero2_2.2.0_ubuntu_focal_desktop_linux4.9.170 HDMI sound works very well.

orangepi-xunlong, 2022-05-19:
Hello, please re-burn the image, because I tested that HDMI audio works, The command used is:

`aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav`

If you open the ` /usr/share/sounds/alsa/audio.wav ` directly on the desktop, HDMI can also hear the sound.

[image]

[image]

agevlakh, 2022-05-19:
@orangepi-xunlong Thanks for the answer. I'll try and write about the result here. Do you have any errors like me in the dmesg and pulseaudio commands?

orangepi-xunlong, 2022-05-19:
Yeah, it's the same here.

agevlakh, 2022-05-19:
@orangepi-xunlong 
Hi, friend! Unfortunately, there was no sound. I changed memory cards, replaced the micro hdmi wire with another one, changed the TV and hdmi connectors, to no avail. Once again, the sound on the board itself is working, but on other Linux images and on Android. I'll try 22.04 when it's available.

orangepi-xunlong, 2022-05-20:
```
aplay /usr/share/sounds/alsa/mute.wav -D hw:1,0 > /dev/null 2>&1
aplay /usr/share/sounds/alsa/mute.wav -D hw:0,0
```

Hi, Please try these two commands to see if there is any sound, thanks!

agevlakh, 2022-05-20:
> 

Hello! Not working... Later I test Ubuntu 22.04 and write about it

agevlakh, 2022-05-20:
On Ubuntu 22.04 xfce also sound breaks for me... 
I'm not an expert, but my observation shows that there are no sound problems on the kernel version 4.x. There is no sound at 5.x. Can I somehow help me find out what is the reason for this behavior for repair?
@orangepi-xunlong

orangepi-xunlong, 2022-05-23:
Hi, how do you test the hdmi audio, thanks!

agevlakh, 2022-05-23:
Hello! I change default sink to HDMI audio. Open volume control, playback menu for monitoring. And do some tests:
- YouTube playing in Firefox
- SoundCloud playing Firefox
- Download video sample and audio samples (mp3, mp4, avi), open it in default audio/video player. Also tested in vlc
- Playing wav files from alsa folder
- Do commands in terminal like in this post
- Changing hdmi connectors, tv, hdmi cable
Hdmi sound works on opi zero 2 with android, Ubuntu 4.9 image, Debian 4.9 image. Also I have raspberry pi 4, also sound works. I can shoot video for you, if you need this

orangepi-xunlong, 2022-05-23:
```
aplay /usr/share/sounds/alsa/audio.wav -D hw:1,0 > /dev/null 2>&1
aplay /usr/share/sounds/alsa/audio.wav -D hw:0,0
```

Please test this and take a screenshot or photo, thanks!

orangepi-xunlong, 2022-05-24:
Please use the image we provide and do not modify anything, thanks!

agevlakh, 2022-05-24:
@orangepi-xunlong I make commands and screenshot on fresh OS image without any modification
[image]

orangepi-xunlong, 2022-05-25:
Looks good this time. If there's no sound, I think we need to find another reason. For example, try a different TV.

Because HDMI sound is fine with me on TV, And I've tested it many times.

You might say that Linux4.9 and Android are fine, Yeah... But I can't guess why it not work on linux5.16, because I can't reproduce the problem here. Thanks!

agevlakh, 2022-05-25:
https://user-images.githubusercontent.com/105754474/170187385-aa950658-c960-4f34-8dbb-3c748eb2615c.mp4

https://user-images.githubusercontent.com/105754474/170187387-e1d330d1-0203-44d7-8846-160c017eeecf.mp4

agevlakh, 2022-05-25:
Different TV no changes situation. I make a video with 5.16 and 4.9. 4.9 have sound
You can download 4.9 image and reproduce it. Why not?

orangepi-xunlong, 2022-05-25:
I can test 4.9 and 5.16 without any problems. Why should I test 4.9???

agevlakh, 2022-05-25:
Apparently, I didn't really understand what you meant in the previous post. In general, I did everything possible to prove that I don't have a sound on 5.16. He also showed that the sound works at 4.9. Unfortunately, I don't know what else to show that the problem is in the firmware and kernel, not in my TV, cable or anything else.

agevlakh, 2022-05-25:
Ok. I test fresh install of Ubuntu and write here about it. Very strange thing

agevlakh, 2022-05-25:
Maybe you have another hardware revision of board? Is it possible? @orangepi-xunlong

agevlakh, 2022-05-25:
And I see difference with our results. I have sound in 4.9 only in this command:
aplay /usr/share/sounds/alsa/audio.wav -D hw:1,0 > /dev/null 2>&1
On this command no have sound:
aplay /usr/share/sounds/alsa/audio.wav -D hw:0,0

But in your video you do only second command and have sound.
Can you explain this? I don't understand. @orangepi-xunlong

orangepi-xunlong, 2022-05-25:
> Do you use image of Ubuntu 22.04 from ORANGEPI site and google drive?

Yes, there is no difference.

> Maybe you have another hardware revision of board? Is it possible?

There is only one version of Zero2, ours will not be different.

> And I see difference with our results. I have sound in 4.9 only in this command:
aplay /usr/share/sounds/alsa/audio.wav -D hw:1,0 > /dev/null 2>&1
On this command no have sound:
aplay /usr/share/sounds/alsa/audio.wav -D hw:0,0

Linux4.9 and Linux5.16 sound card numbers are different.

For linux4.9, The playback command is: 

```
aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav
```

For linux5.16, The playback command is: 

```
aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav
```

> But in your video you do only second command and have sound. First command you not test.
> Can you explain this? I don't understand. @orangepi-xunlong

Because the first command is to initialize the HDMI audio device, it only needs to be executed once.

agevlakh, 2022-05-25:
Hello again friend @orangepi-xunlong ! I want to announce the news. I got a monitor with sound output and hdmi. It has sound at 5.16. Then there may be an assumption that something is wrong with my TV. But everything is fine with him, because he plays sound through hdmi with 4.19 linux, with android, as well as from other devices (laptop, computer, raspberry pi). I also used the same HDMI cable to connect to the monitor and TV. A friend from our local forum confirmed that he had the same problem as me, but he did not say which device he connected to his zero 2. Judging by your video with working sound, you connect your zero 2 to the monitor. It will not make it difficult for you to try to test it on a TV, preferably a Samsung. Or what is available. I only have 2 Samsung TVs, my problem is relevant on them. Thank you!

agevlakh, 2022-05-25:
As you explained to me, in the 4.19 kernel, sound is output differently than in the 5.16 kernel. Can you tell me if the Orange Pi team can build a 5.16 kernel for tests and publish it, which will output sound like a 4.19 kernel for experiments? Thank you!

orangepi-xunlong, 2022-05-26:
> Judging by your video with working sound, you connect your zero 2 to the monitor.  It will not make it difficult for you to try to test it on a TV, preferably a Samsung.

No, It is a Philips TV in the orangepizero2_linux5.16_hdmi_audio_test.mp4. I don't have a Samsung TV, so I can't test it.

> As you explained to me, in the 4.9 kernel, sound is output differently than in the 5.16 kernel. 

This is just a difference in the number of the order.  It doesn't really affect the audio playback.

agevlakh, 2022-05-26:
> It doesn't really affect the audio playback.

How does it not affect? I'm sorry, but what have we been doing here for the last few days? If it didn't, I wouldn't have opened this bug. And you will agree that if it works for you, it does not mean that it works for everyone, especially since I have proved the inoperable on the new core. Plus 2 people from the forum confirmed that there was a similar problem.

orangepi-xunlong, 2022-05-26:
I don't want to explain this to you anymore, even if some TV has a problem, BUT I'm not going to solve this problem, because I can't solve it...

closed 2022-05-26 by orangepi-xunlong

agevlakh, 2022-05-26:
then invite a more competent person to this ticket who can help, as this is a real problem.

djoiner22, 2024-05-14:
> I don't want to explain this to you anymore, even if some TV has a problem, BUT I'm not going to solve this problem, because I can't solve it...

> I don't want to explain this to you anymore, even if some TV has a problem, BUT I'm not going to solve this problem, because I can't solve it...

people will stop buying your products if you dont solve issues like this and you provide shitty support. I am having the same issue and as tech support for a multi-billion dolar company I would have lost my job for such an inadequate response to a customer.