orangepi-xunlong/orangepi-build #34: Cant launch chromium on Orange Pi Zero 2 Debian XFCE 3.0.0

agevlakh, 2022-05-18:
Hello! I use image Orangepizero2_3.0.0_debian_bullseye_desktop_xfce_linux5.16.17 and I cant run chromium. I install it, but it not launch at all. If I launch chromium in terminal I see this output:
```
orangepi@orangepizero2:~$ chromium 
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2777:2777:0518/215116.453094:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2777:2815:0518/215116.481654:ERROR:zygote_host_impl_linux.cc(262)] Failed to adjust OOM score of renderer with pid 2842: Permission denied (13)
[2777:2777:0518/215116.962688:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=6
[2777:2843:0518/215116.973937:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[2777:2843:0518/215116.983793:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2777:2843:0518/215116.991157:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
*** stack smashing detected ***: terminated
[2777:2777:0518/215117.039239:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2777:2777:0518/215117.114073:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2777:2777:0518/215117.306994:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=6
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
*** stack smashing detected ***: terminated
[2777:2777:0518/215117.377913:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=6
[2777:2777:0518/215117.378058:FATAL:gpu_data_manager_impl_private.cc(447)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap
```
I launch chromium with argument --disable-gpu, --disable-gpu-driver-bug-workarounds, --disable-features=UseChromeOSDirectVideoDecoder , but it not help and I saw same log.
Also I try to install mesa-vdpau-drivers, libvdpau-va-gl1, va-driver-all libva2, libva-glx2 and I have same issue.
Also I tyr update graphics drivers like mesa and OpenGL components from sid repository, but it not help.
On Orangepizero2_2.2.0_ubuntu_focal_desktop_linux4.9.170 image chromium browser installed by default and works.
Please help solve this issue. Thank you!

orangepi-xunlong, 2022-05-19:
Hello , 

Yes, chromium isn't available in Debian11, so I pre-installed firefox by default.

But chromium is ok in Debian12 and Ubuntu22.04.

[image]

orangepi-xunlong, 2022-05-19:
Ubuntu22.04 will be uploaded tomorrow. Debian12 can be compiled using orangepi-build next branch.

https://github.com/orangepi-xunlong/orangepi-build/tree/next

Thanks.

agevlakh, 2022-05-19:
@orangepi-xunlong Thanks for the answer!

closed 2022-05-23 by orangepi-xunlong