orangepi-xunlong/orangepi-build #35: Orange Pi 3 LTS  HDMI & 3.5 mm audio no Sound with Orangepi3-lts_3.0.0_debian_bullseye_desktop_xfce_linux5.16.17

kaosheng, 2022-06-07:
Orange Pi 3 LTS  HDMI & 3.5 mm audio no Sound with  
Offical Download : Orangepi3-lts_3.0.0_debian_bullseye_desktop_xfce_linux5.16.17
HDMI  no Sound
Offical Download: Orangepi3-lts_2.2.2_debian_buster_desktop_linux5.10.75.
HDMI No Sound
ArmBian Bullsye:
HDMI No Sound
Bluetooth Cant use

3.5mm sometimes  OK But Often NG

orangepi-xunlong, 2022-06-07:
hdmi test:

`aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav`

3.5 mm audio test:

` aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav`

blackmeat1225, 2022-06-07:
orangepi@orangepi3-lts:~$ aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav
aplay: main:830: audio open error: 裝置或系統資源忙碌中

orangepi@orangepi3-lts:~$  aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav
aplay: main:830: audio open error: 裝置或系統資源忙碌中

orangepi@orangepi3-lts:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: allwinnerac200c [allwinner,ac200-codec], device 0: 508f000.i2s-acx00-dai acx00-codec-0 [508f000.i2s-acx00-dai acx00-codec-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 1: sun9ihdmi [sun9i-hdmi], device 0: SUN9I-HDMI PCM i2s-hifi-0 [SUN9I-HDMI PCM i2s-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

orangepi@orangepi3-lts:~$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default
    Playback/recording through the PulseAudio sound server
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
hw:CARD=allwinnerac200c,DEV=0
    allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
    Direct hardware device without any conversions
plughw:CARD=allwinnerac200c,DEV=0
    allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
    Hardware device with all software conversions
sysdefault:CARD=allwinnerac200c
    allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
    Default Audio Device
dmix:CARD=allwinnerac200c,DEV=0
    allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
    Direct sample mixing device
usbstream:CARD=allwinnerac200c
    allwinner,ac200-codec
    USB Stream Output
hw:CARD=sun9ihdmi,DEV=0
    sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
    Direct hardware device without any conversions
plughw:CARD=sun9ihdmi,DEV=0
    sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
    Hardware device with all software conversions
sysdefault:CARD=sun9ihdmi
    sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
    Default Audio Device
dmix:CARD=sun9ihdmi,DEV=0
    sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
    Direct sample mixing device
usbstream:CARD=sun9ihdmi
    sun9i-hdmi
    USB Stream Output

orangepi@orangepi3-lts:~$ uname -a
Linux orangepi3-lts 5.16.17-sun50iw6 #3.0.0 SMP Tue Apr 19 10:19:38 CST 2022 aarch64 GNU/Li

kaosheng, 2022-06-07:
> orangepi@orangepi3-lts:~$ aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav aplay: main:830: audio open error: 裝置或系統資源忙碌中
>
> orangepi@orangepi3-lts:~$ aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav [...]

orangepi@orangepi3-lts:~$ aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav
aplay: main:830: audio open error: 裝置或系統資源忙碌中

orangepi@orangepi3-lts:~$ aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav
aplay: main:830: audio open error: 裝置或系統資源忙碌中

orangepi@orangepi3-lts:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: allwinnerac200c [allwinner,ac200-codec], device 0: 508f000.i2s-acx00-dai acx00-codec-0 [508f000.i2s-acx00-dai acx00-codec-0]
Subdevices: 0/1
Subdevice #0: subdevice #0
card 1: sun9ihdmi [sun9i-hdmi], device 0: SUN9I-HDMI PCM i2s-hifi-0 [SUN9I-HDMI PCM i2s-hifi-0]
Subdevices: 0/1
Subdevice #0: subdevice #0

orangepi@orangepi3-lts:~$ aplay -L
null
Discard all samples (playback) or generate zero samples (capture)
default
Playback/recording through the PulseAudio sound server
lavrate
Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
Rate Converter Plugin Using Samplerate Library
speexrate
Rate Converter Plugin Using Speex Resampler
jack
JACK Audio Connection Kit
oss
Open Sound System
pulse
PulseAudio Sound Server
upmix
Plugin for channel upmix (4,6,8)
vdownmix
Plugin for channel downmix (stereo) with a simple spacialization
hw:CARD=allwinnerac200c,DEV=0
allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
Direct hardware device without any conversions
plughw:CARD=allwinnerac200c,DEV=0
allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
Hardware device with all software conversions
sysdefault:CARD=allwinnerac200c
allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
Default Audio Device
dmix:CARD=allwinnerac200c,DEV=0
allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
Direct sample mixing device
usbstream:CARD=allwinnerac200c
allwinner,ac200-codec
USB Stream Output
hw:CARD=sun9ihdmi,DEV=0
sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
Direct hardware device without any conversions
plughw:CARD=sun9ihdmi,DEV=0
sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
Hardware device with all software conversions
sysdefault:CARD=sun9ihdmi
sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
Default Audio Device
dmix:CARD=sun9ihdmi,DEV=0
sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
Direct sample mixing device
usbstream:CARD=sun9ihdmi
sun9i-hdmi
USB Stream Output

orangepi@orangepi3-lts:~$ uname -a
Linux orangepi3-lts 5.16.17-sun50iw6 https://github.com/orangepi-xunlong/orangepi-build/issues/3.0.0 SMP Tue Apr 19 10:19:38 CST 2022 aarch64 GNU/Li

kaosheng, 2022-06-07:
> hdmi test:
> 
> `aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav`
> 
> 3.5 mm audio test:
> 
> ` aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav`
orangepi@orangepi3-lts:~$ aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav
aplay: main:830: audio open error: 裝置或系統資源忙碌中

orangepi@orangepi3-lts:~$ aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav
aplay: main:830: audio open error: 裝置或系統資源忙碌中

orangepi@orangepi3-lts:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: allwinnerac200c [allwinner,ac200-codec], device 0: 508f000.i2s-acx00-dai acx00-codec-0 [508f000.i2s-acx00-dai acx00-codec-0]
Subdevices: 0/1
Subdevice #0: subdevice #0
card 1: sun9ihdmi [sun9i-hdmi], device 0: SUN9I-HDMI PCM i2s-hifi-0 [SUN9I-HDMI PCM i2s-hifi-0]
Subdevices: 0/1
Subdevice #0: subdevice #0

orangepi@orangepi3-lts:~$ aplay -L
null
Discard all samples (playback) or generate zero samples (capture)
default
Playback/recording through the PulseAudio sound server
lavrate
Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
Rate Converter Plugin Using Samplerate Library
speexrate
Rate Converter Plugin Using Speex Resampler
jack
JACK Audio Connection Kit
oss
Open Sound System
pulse
PulseAudio Sound Server
upmix
Plugin for channel upmix (4,6,8)
vdownmix
Plugin for channel downmix (stereo) with a simple spacialization
hw:CARD=allwinnerac200c,DEV=0
allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
Direct hardware device without any conversions
plughw:CARD=allwinnerac200c,DEV=0
allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
Hardware device with all software conversions
sysdefault:CARD=allwinnerac200c
allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
Default Audio Device
dmix:CARD=allwinnerac200c,DEV=0
allwinner,ac200-codec, 508f000.i2s-acx00-dai acx00-codec-0
Direct sample mixing device
usbstream:CARD=allwinnerac200c
allwinner,ac200-codec
USB Stream Output
hw:CARD=sun9ihdmi,DEV=0
sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
Direct hardware device without any conversions
plughw:CARD=sun9ihdmi,DEV=0
sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
Hardware device with all software conversions
sysdefault:CARD=sun9ihdmi
sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
Default Audio Device
dmix:CARD=sun9ihdmi,DEV=0
sun9i-hdmi, SUN9I-HDMI PCM i2s-hifi-0
Direct sample mixing device
usbstream:CARD=sun9ihdmi
sun9i-hdmi
USB Stream Output

orangepi@orangepi3-lts:~$ uname -a
Linux orangepi3-lts 5.16.17-sun50iw6 https://github.com/orangepi-xunlong/orangepi-build/issues/3.0.0 SMP Tue Apr 19 10:19:38 CST 2022 aarch64 GNU/Li

orangepi-xunlong, 2022-06-13:
I retested it and it's fine.
[image]

kaosheng, 2022-06-17:
> I retested it and it's fine. [image]

I tried again,i found out when i use orange pi config to change locale to ZH_TW UT8,THE PROBLEM COMING,NO sound & youtube stock

orangepi-xunlong, 2022-06-17:
使用youtube播放视频卡顿的话是正常的，因为linux系统没有视频硬件解码功能。

kaosheng, 2022-06-17:
> 使用youtube播放视频卡顿的话是正常的，因为linux系统没有视频硬件解码功能。

不是卡頓是無法撥放，因為音效的問題無法撥放，當音效變成了dummy output,時，YouTube就可以撥放，但沒有聲音

orangepi-xunlong, 2022-06-17:
hdmi test:

aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav

3.5 mm audio test:

aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav

先看下这个命令有没有声音

kaosheng, 2022-06-17:
> hdmi test:
> 
> aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav
> 
> 3.5 mm audio test:
> 
> aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav
> 
> 先看下这个命令有没有声音

有聲音，正常，
但是orangepi config,變更語言後，就壞掉了
aplay: main:830: audio open error: 裝置或系統資源忙碌中

kaosheng, 2022-06-21:
> hdmi test:
> 
> aplay -D hw:1,0 /usr/share/sounds/alsa/audio.wav
> 
> 3.5 mm audio test:
> 
> aplay -D hw:0,0 /usr/share/sounds/alsa/audio.wav
> 
> 先看下这个命令有没有声音

不知道是發生了什麼問題？3.5mm & HDMI無法使用
https://youtu.be/oXxSWJPSeCU

kaosheng, 2022-06-29:
https://github.com/home-assistant/plugin-audio/pkgs/container/aarch64-hassio-audio#container
The hassio_audio makes system no audio

closed 2022-06-29 by kaosheng