orangepi-xunlong/orangepi-build #36: Build U-boot -  multiple definition of `yylloc'

kmlebedev, 2022-06-10:
```
root@35b6a4c971d9:~/OrangePiRK3399# ./build.sh
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
automake is already the newest version (1:1.16.5-1.3).
bc is already the newest version (1.07.1-3build1).
dosfstools is already the newest version (4.2-1build3).
gcc is already the newest version (4:11.2.0-1ubuntu1).
lib32z1 is already the newest version (1:1.2.11.dfsg-2ubuntu9).
lib32z1-dev is already the newest version (1:1.2.11.dfsg-2ubuntu9).
make is already the newest version (4.3-4.1build1).
mtools is already the newest version (4.0.33-1+really4.0.32-1build1).
pv is already the newest version (1.6.6-1build2).
u-boot-tools is already the newest version (2022.01+dfsg-2ubuntu2).
libarchive-tools is already the newest version (3.6.0-1ubuntu1).
python2-minimal is already the newest version (2.7.18-3).
debootstrap is already the newest version (1.0.126+nmu1ubuntu0.1).
qemu-user-static is already the newest version (1:6.2+dfsg-2ubuntu6.1).
The following additional packages will be installed:
  libexpat1-dev libncurses-dev libpipeline1 libpython2.7 swig4.0
Suggested packages:
  bison-doc build-essential flex-doc ncurses-doc libssl-doc swig-doc swig-examples swig4.0-examples swig4.0-doc
Recommended packages:
  libfl-dev
The following NEW packages will be installed:
  binfmt-support bison dos2unix flex libexpat1-dev libncurses-dev libncurses5-dev libpipeline1 libpython2.7 libpython2.7-dev libssl-dev swig swig4.0
0 upgraded, 13 newly installed, 0 to remove and 5 not upgraded.
Need to get 9208 kB of archives.
After this operation, 42.5 MB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu jammy/main amd64 flex amd64 2.6.4-8build2 [307 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy/main amd64 libpipeline1 amd64 1.5.5-1 [23.5 kB]
Get:3 http://archive.ubuntu.com/ubuntu jammy/main amd64 binfmt-support amd64 2.2.1-2 [55.8 kB]
Get:4 http://archive.ubuntu.com/ubuntu jammy/main amd64 bison amd64 2:3.8.2+dfsg-1build1 [748 kB]
Get:5 http://archive.ubuntu.com/ubuntu jammy/main amd64 libexpat1-dev amd64 2.4.7-1 [147 kB]
Get:6 http://archive.ubuntu.com/ubuntu jammy/main amd64 libncurses-dev amd64 6.3-2 [380 kB]
Get:7 http://archive.ubuntu.com/ubuntu jammy/main amd64 libncurses5-dev amd64 6.3-2 [780 B]
Get:8 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libpython2.7 amd64 2.7.18-13ubuntu1 [1162 kB]
Get:9 http://archive.ubuntu.com/ubuntu jammy/universe amd64 libpython2.7-dev amd64 2.7.18-13ubuntu1 [2512 kB]
Get:10 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 libssl-dev amd64 3.0.2-0ubuntu1.2 [2372 kB]
Get:11 http://archive.ubuntu.com/ubuntu jammy/universe amd64 swig4.0 amd64 4.0.2-1ubuntu1 [1110 kB]
Get:12 http://archive.ubuntu.com/ubuntu jammy/universe amd64 swig all 4.0.2-1ubuntu1 [5632 B]
Get:13 http://archive.ubuntu.com/ubuntu jammy/universe amd64 dos2unix amd64 7.4.2-2 [384 kB]
Fetched 9208 kB in 2s (4170 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package flex.
(Reading database ... 15559 files and directories currently installed.)
Preparing to unpack .../00-flex_2.6.4-8build2_amd64.deb ...
Unpacking flex (2.6.4-8build2) ...
Selecting previously unselected package libpipeline1:amd64.
Preparing to unpack .../01-libpipeline1_1.5.5-1_amd64.deb ...
Unpacking libpipeline1:amd64 (1.5.5-1) ...
Selecting previously unselected package binfmt-support.
Preparing to unpack .../02-binfmt-support_2.2.1-2_amd64.deb ...
Unpacking binfmt-support (2.2.1-2) ...
Selecting previously unselected package bison.
Preparing to unpack .../03-bison_2%3a3.8.2+dfsg-1build1_amd64.deb ...
Unpacking bison (2:3.8.2+dfsg-1build1) ...
Selecting previously unselected package libexpat1-dev:amd64.
Preparing to unpack .../04-libexpat1-dev_2.4.7-1_amd64.deb ...
Unpacking libexpat1-dev:amd64 (2.4.7-1) ...
Selecting previously unselected package libncurses-dev:amd64.
Preparing to unpack .../05-libncurses-dev_6.3-2_amd64.deb ...
Unpacking libncurses-dev:amd64 (6.3-2) ...
Selecting previously unselected package libncurses5-dev:amd64.
Preparing to unpack .../06-libncurses5-dev_6.3-2_amd64.deb ...
Unpacking libncurses5-dev:amd64 (6.3-2) ...
Selecting previously unselected package libpython2.7:amd64.
Preparing to unpack .../07-libpython2.7_2.7.18-13ubuntu1_amd64.deb ...
Unpacking libpython2.7:amd64 (2.7.18-13ubuntu1) ...
Selecting previously unselected package libpython2.7-dev:amd64.
Preparing to unpack .../08-libpython2.7-dev_2.7.18-13ubuntu1_amd64.deb ...
Unpacking libpython2.7-dev:amd64 (2.7.18-13ubuntu1) ...
Selecting previously unselected package libssl-dev:amd64.
Preparing to unpack .../09-libssl-dev_3.0.2-0ubuntu1.2_amd64.deb ...
Unpacking libssl-dev:amd64 (3.0.2-0ubuntu1.2) ...
Selecting previously unselected package swig4.0.
Preparing to unpack .../10-swig4.0_4.0.2-1ubuntu1_amd64.deb ...
Unpacking swig4.0 (4.0.2-1ubuntu1) ...
Selecting previously unselected package swig.
Preparing to unpack .../11-swig_4.0.2-1ubuntu1_all.deb ...
Unpacking swig (4.0.2-1ubuntu1) ...
Selecting previously unselected package dos2unix.
Preparing to unpack .../12-dos2unix_7.4.2-2_amd64.deb ...
Unpacking dos2unix (7.4.2-2) ...
Setting up libpipeline1:amd64 (1.5.5-1) ...
Setting up libpython2.7:amd64 (2.7.18-13ubuntu1) ...
Setting up flex (2.6.4-8build2) ...
Setting up libncurses-dev:amd64 (6.3-2) ...
Setting up libexpat1-dev:amd64 (2.4.7-1) ...
Setting up libssl-dev:amd64 (3.0.2-0ubuntu1.2) ...
Setting up binfmt-support (2.2.1-2) ...
invoke-rc.d: could not determine current runlevel
invoke-rc.d: policy-rc.d denied execution of restart.
Setting up libncurses5-dev:amd64 (6.3-2) ...
Setting up bison (2:3.8.2+dfsg-1build1) ...
update-alternatives: using /usr/bin/bison.yacc to provide /usr/bin/yacc (yacc) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/man1/yacc.1.gz because associated file /usr/share/man/man1/bison.yacc.1.gz (of link group yacc) doesn't exist
Setting up dos2unix (7.4.2-2) ...
Setting up swig4.0 (4.0.2-1ubuntu1) ...
Setting up libpython2.7-dev:amd64 (2.7.18-13ubuntu1) ...
Setting up swig (4.0.2-1ubuntu1) ...
Processing triggers for libc-bin (2.35-0ubuntu3) ...
 Build U-boot
make for rk3399_defconfig by -j8
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config.h
  UPD     include/config.h
  CFG     u-boot.cfg
  GEN     include/autoconf.mk.dep
  CFG     tpl/u-boot.cfg
  CFG     spl/u-boot.cfg
  GEN     include/autoconf.mk
  GEN     spl/include/autoconf.mk
  GEN     tpl/include/autoconf.mk
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  SHIPPED scripts/dtc/dtc-lexer.lex.c
  SHIPPED scripts/dtc/dtc-parser.tab.h
  SHIPPED scripts/dtc/dtc-parser.tab.c
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  CHK     include/config.h
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  CFG     u-boot.cfg
  HOSTLD  scripts/dtc/dtc
/usr/bin/ld: scripts/dtc/dtc-parser.tab.o:(.bss+0x10): multiple definition of `yylloc'; scripts/dtc/dtc-lexer.lex.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [scripts/Makefile.host:108: scripts/dtc/dtc] Error 1
make[1]: *** [scripts/Makefile.build:425: scripts/dtc] Error 2
make: *** [Makefile:491: scripts] Error 2
make: *** Waiting for unfinished jobs....
  UPD     include/config/uboot.release
```

orangepi-xunlong, 2022-06-13:
Hi, Are you using orangepi-build? Which board?

kmlebedev, 2022-06-14:
`make: Leaving directory '/root/OrangePiRK3399/kernel'
 Complete kernel compilation ...
Destination /root/OrangePiRK3399/output/rootfs not found or not a directory.
Create /root/OrangePiRK3399/output/rootfs
Downloading bionic rootfs tarball ...
--2022-06-10 13:12:41--  https://mirrors.tuna.tsinghua.edu.cn/ubuntu-cdimage/ubuntu-base/releases/bionic/release/ubuntu-base-18.04-base-arm64.tar.gz
Resolving mirrors.tuna.tsinghua.edu.cn (mirrors.tuna.tsinghua.edu.cn)... 101.6.15.130, 2402:f000:1:400::2
Connecting to mirrors.tuna.tsinghua.edu.cn (mirrors.tuna.tsinghua.edu.cn)|101.6.15.130|:443... connected.`

kmlebedev, 2022-06-23:
> Hi, Are you using orangepi-build? Which board?

Yes, I realized that this project for assembling a legacy image was abandoned and switched to armbian
https://forum.armbian.com/topic/3946-rk3399-orange-pi/page/7/#comment-142192

There's an advanced imaging framework, why don't you reuse it for new boards?

kmlebedev, 2022-06-24:
At some point, the board stopped booting from the SD card, even though I was running install_to_emmc
```
cdn-dp fec00000.dp: Direct firmware load for rockchip/dptx.bin Failed with error -2
xhci-hcd xhci-hcd.11.auto: Time out while waiting for setup device commnad
```