orangepi-xunlong/orangepi-build #38: gstreamer only for Desktop?

AndreiCherniaev, 2022-06-21:
Hello.

I build OS for server, not Desktop. I use
./build.sh  BOARD=orangepi4 BRANCH=legacy BUILD_OPT=image RELEASE=focal BUILD_MINIMAL=yes **BUILD_DESKTOP=no** KERNEL_ONLY=no KERNEL_CONFIGURE=yes  COMPRESS_OUTPUTIMAGE=img OFFLINE_WORK=yes OPI_USERNAME=a  OPI_PWD=0 CREATE_PATCHES=yes SERIALCON=none

But build script try to add some of [gstreamer](https://github.com/orangepi-xunlong/rk3399_gst_xserver_libs). There are a lot of errors while building:
_[ .... ] Installing [ orangepi-firmware_2.2.2_all.deb ]
cp: cannot create regular file '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/share/sounds/alsa/': No such file or directory
cp: cannot stat '/home/a/Documents/XunlongA/external/cache/sources/rk3399_gst_xserver_libs/gstreamer/focal_build/etc/iqfiles': No such file or directory
cp: missing destination file operand after '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/bin/'
Try 'cp --help' for more information.
cp: missing destination file operand after '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/include/'
Try 'cp --help' for more information.
cp: missing destination file operand after '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/lib/'
Try 'cp --help' for more information.
cp: missing destination file operand after '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/share/'
Try 'cp --help' for more information.
cp: missing destination file operand after '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/bin/'
Try 'cp --help' for more information.
cp: missing destination file operand after '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/include/'
Try 'cp --help' for more information.
cp: missing destination file operand after '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/lib/'
Try 'cp --help' for more information.
cp: missing destination file operand after '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/share/'
Try 'cp --help' for more information.
cp: missing destination file operand after '/home/a/Documents/XunlongA/.tmp/rootfs-legacy-orangepi4-focal-no-yes/usr/libexec/'
Try 'cp --help' for more information.
[ o.k. ] Calling image customization script [ customize-image.sh ]_

May be Xunlong should check no only
if [[ $BOARD == orangepi4 ]]; then
But also
if [[ $BUILD_DESKTOP == no ]]; then
?

https://github.com/orangepi-xunlong/orangepi-build/blob/9d1c8c9c4a86194eacd09a422da8985e0175eba0/scripts/main.sh#L394

orangepi-xunlong, 2022-06-23:
Hi, is the main branch code used or next branch？

AndreiCherniaev, 2022-06-24:
> in br

I use main branch. I change RELEASE to bionic and now no error.

closed 2022-06-27 by orangepi-xunlong