orangepi-xunlong/orangepi-build #45: [ error ] ERROR in function compile_uboot [ build.sh:305 -> scripts/main.sh:596 -> scripts/main.sh:479 -> scripts/compilation.sh:242 -> scripts/general.sh:0 ]

yuta-sakata, 2022-09-17:
asus@asus-1:~/project/orangepi/orangepi-build$ ./build.sh 
[ warn ] This script requires root privileges, trying to use sudo 
[ o.k. ] Using config file [ /home/asus/project/orangepi/orangepi-build/userpatches/config-example.conf ]
[ .... ] Extension being added [ sunxi-tools :: added by build.sh:305 -> scripts/main.sh:358 -> scripts/configuration.sh:150 -> external/config/sources/families/sun50iw6.conf:2 -> external/config/sources/families/include/sunxi64_common.inc:1 -> scripts/extensions.sh:0 ]
[ o.k. ] Extension manager [ processed 3 Extension Methods calls and 3 Extension Method implementations ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ jammy ]
[ .... ] Installing build dependencies 
[ o.k. ] Syncing clock [ cn.pool.ntp.org ]
[ o.k. ] Checking for external GCC compilers 
[ o.k. ] Downloading sources 
[ o.k. ] Checking git sources [ /home/asus/project/orangepi/orangepi-build/u-boot v2021.10-sunxi ]
[ .... ] Creating local copy 
[ .... ] Fetching updates 
remote: Enumerating objects: 29850, done.
remote: Counting objects: 100% (10203/10203), done.
remote: Compressing objects: 100% (6240/6240), done.
remote: Total 29850 (delta 4413), reused 3974 (delta 3963), pack-reused 19647
接收对象中: 100% (29850/29850), 36.55 MiB | 9.46 MiB/s, 完成.
处理 delta 中: 100% (10069/10069), 完成.
来自 https://github.com/orangepi-xunlong/u-boot-orangepi
 * branch              v2021.10-sunxi -> FETCH_HEAD
 * [新分支]            v2021.10-sunxi -> origin/v2021.10-sunxi
[ .... ] Checking out 
[ o.k. ] Checking git sources [ /home/asus/project/orangepi/orangepi-build/kernel orange-pi-5.16-sunxi64 ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ /home/asus/project/orangepi/orangepi-build/external/cache/sources/arm-trusted-firmware-sunxi-mainline v2.2 ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ /home/asus/project/orangepi/orangepi-build/external/cache/sources/sunxi-tools master ]
[ .... ] Up to date 
[ o.k. ] Cleaning /home/asus/project/orangepi/orangepi-build/output/debs for [ orangepi3 next ]
[ o.k. ] Compiling ATF 
[ o.k. ] Compiler version [ aarch64-none-linux-gnu-gcc 9.2.1 ]
[ o.k. ] Started patching process for [ atf sun50iw6-orangepi3-next ]
[ o.k. ] Looking for user patches in [ userpatches/atf/atf-sun50iw6 ]
[ o.k. ] Compiling u-boot [ v2021.10 ]
[ o.k. ] Compiler version [ aarch64-none-linux-gnu-gcc 9.2.1 ]
[ o.k. ] Started patching process for [ u-boot sun50iw6-orangepi3-next ]
[ o.k. ] Looking for user patches in [ userpatches/u-boot/u-boot-sunxi ]
'/tmp/tmp.Izp7PIpLSY/bl31.bin' -> './bl31.bin'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
scripts/kconfig/conf  --syncconfig Kconfig
  UPD     include/config.h
  CFG     u-boot.cfg
  GEN     include/autoconf.mk.dep
  CFG     spl/u-boot.cfg
  GEN     include/autoconf.mk
  GEN     spl/include/autoconf.mk
  UPD     include/generated/dt.h
  UPD     include/generated/timestamp_autogenerated.h
  UPD     include/config/uboot.release
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  SHIPPED scripts/dtc/pylibfdt/libfdt.i
  LEX     scripts/dtc/dtc-lexer.lex.c
[... 475 lines cut ...]
  AR      lib/libfdt/built-in.o
  AS      lib/efi_loader/efi_crt0.o
  CC      lib/efi_loader/efi_reloc.o
  CC      lib/slre.o
  CC      lib/efi_loader/efi_freestanding.o
  CC      lib/string.o
  CC      lib/tables_csum.o
  CC      lib/time.o
  CC      lib/hexdump.o
  CC      lib/uuid.o
  CC      lib/rand.o
  CC      lib/panic.o
  AR      lib/efi_loader/built-in.o
  CC      lib/vsprintf.o
  LD      lib/efi_loader/helloworld_efi.so
  CC      lib/strto.o
  CC      lib/date.o
  CC      lib/rtc-lib.o
  CC      lib/elf.o
  OBJCOPY lib/efi_loader/helloworld.efi
  AR      lib/built-in.o
  LD      u-boot
[ error ] ERROR in function compile_uboot [ build.sh:305 -> scripts/main.sh:596 -> scripts/main.sh:479 -> scripts/compilation.sh:242 -> scripts/general.sh:0 ]
[ error ] U-boot compilation failed 
[ o.k. ] Process terminated

yuta-sakata, 2022-09-17:
My target board is Orange pi 3

yuta-sakata, 2022-09-17:
==  atf  ==

	== u-boot make orangepi_3_defconfig ==

	== u-boot make  ==

.config:1612:warning: override: reassigning to symbol BOOTDELAY
arch/arm/mach-sunxi/pmic_bus.c: 在函数‘pmic_bus_init’中:
arch/arm/mach-sunxi/pmic_bus.c:48:2: 警告： 隐式声明函数‘clock_twi_onoff’ [-Wimplicit-function-declaration]
   48 |  clock_twi_onoff(5, 1);
      |  ^~~~~~~~~~~~~~~
drivers/power/axp806.c: 在函数‘axp_init’中:
drivers/power/axp806.c:78:12: 警告： 未使用的变量‘val’ [-Wunused-variable]
   78 |         u8 val;
      |            ^~~
/home/asus/project/orangepi/orangepi-build/toolchains/gcc-arm-9.2-2019.12-x86_64-aarch64-none-linux-gnu/bin/aarch64-none-linux-gnu-ld.bfd: arch/arm/mach-sunxi/pmic_bus.o: in function `pmic_bus_write':
/home/asus/project/orangepi/orangepi-build/u-boot/v2021.10-sunxi/arch/arm/mach-sunxi/pmic_bus.c:99: undefined reference to `i2c_write'
Segmentation fault (core dumped)
make: *** [Makefile:1790：u-boot] 错误 139
make: *** 正在删除文件“u-boot”

nairn62, 2022-12-18:
I created a patch for this issue: u-boot-sun50iw6-next.patch
Place patch in folder: orangepi-build/output/patch

diff --git a/configs/orangepi_3_defconfig b/configs/orangepi_3_defconfig
index d82b8082..e9e375d4 100644
--- a/configs/orangepi_3_defconfig
+++ b/configs/orangepi_3_defconfig
@@ -11,6 +11,8 @@ CONFIG_HDMI_DDC_EN="PH2"
 CONFIG_DEFAULT_DEVICE_TREE="sun50i-h6-orangepi-3"
 CONFIG_SPL=y
+CONFIG_SPL_I2C=y
+CONFIG_R_I2C_ENABLE=y

nairn62, 2022-12-18:
[u-boot-sun50iw6-next.patch](https://github.com/orangepi-xunlong/orangepi-build/files/10253696/u-boot-sun50iw6-next.patch)

Patch attached should be placed in the following folder (you might have to create the path yourself):
userpatches/u-boot/u-boot-sunxi/

1696774082, 2024-01-31:
+1 我这个是orangepi zero2 [ error ] ERROR in function compile_uboot [ ./build.sh:305 -> scripts/main.sh:630 -> scripts/main.sh:516 -> scripts/compilation.sh:256 -> scripts/general.sh:0 ]
[ error ] U-boot compilation failed

PureBulb, 2024-05-06:
+1 本来用docker搭建一个流程下来就这里出了问题我以为是docker的问题结果用官方镜像推荐的22.24也是这个样子

PureBulb, 2024-05-06:
> I created a patch for this issue: u-boot-sun50iw6-next.patch
> 
> [u-boot-sun50iw6-next.patch](https://github.com/orangepi-xunlong/orangepi-build/files/10253696/u-boot-sun50iw6-next.patch)
> 
> The patch attached should be placed in the following folder (you might have to create the path yourself): userpatches/u-boot/u-boot-sunxi/

thank you for your help,i have already solved this problem

KirsminX, 2024-11-17:
本人使用的是 Zero3，遇到同样的问题。使用其他用户发布的补丁没有效果 [u-boot-sun50iw6-next.patch](https://github.com/orangepi-xunlong/orangepi-build/files/10253696/u-boot-sun50iw6-next.patch)

gxk2000, 2025-03-24:
yes！it work！ thank you for your help！