orangepi-xunlong/orangepi-build #53: PCIe fail to initialize on Orange PI 5

fenrirjk, 2022-12-26:
Hi,

I received an Orange Pi 5 and I could not use the NVMe SSD.

I tried on Orangepi5_1.0.2_debian_bullseye_server_linux5.10.110.7z and Orangepi5_1.0.6_ubuntu_focal_server_linux5.10.110.7z with the same result.

After some investigation I noticed that tthe PCIe fails to initialize with:
```
[    8.992510] rk-pcie fe190000.pcie: PCIe Link Fail
[    8.992526] rk-pcie fe190000.pcie: failed to initialize host
```

This happens with or without the NVMe SSD plugged in.

See below for the full dmesg log: https://pastebin.com/4UZ03yAP

orangepi-xunlong, 2023-01-03:
NVMe SSD is ok for me.  No additional settings are required.

fenrirjk, 2023-01-03:
> NVMe SSD is ok for me. No additional settings are required.

Does it mean I have a hardware failure on my board ?

fenrirjk, 2023-01-04:
> I can only tell you that the software supports NVME SSD.
> 
> Whether there is a problem with the hardware, I don't know. You can try it with another NVME SSD.

Since PCIe bus cannot initializes even **without anything plugged in the M.2 slot**, I don't think using another NVME SSD (that I don't have and I don't want to buy) would help.

IMO, the PCIe bus should first initialize correctly before further testing.

danielpinto8zz6, 2023-01-04:
> Hi,
> 
> I received an Orange Pi 5 and I could not use the NVMe SSD.
> 
> I tried with
> 
> * Orangepi5_1.0.2_debian_bullseye_server_linux5.10.110.7z
> * Orangepi5_1.0.6_ubuntu_focal_server_linux5.10.110.7z with the same result.
> 
> After some investigation I noticed that tthe PCIe fails to initialize with:
> 
> ```
> $ cat dmesg.txt | grep rk-pcie
> [    6.086227] rk-pcie fe190000.pcie: invalid prsnt-gpios property in node
> [    6.086238] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply from device tree
> [    6.086922] rk-pcie fe190000.pcie: missing legacy IRQ resource
> [    6.086939] rk-pcie fe190000.pcie: IRQ msi not found
> [    6.086945] rk-pcie fe190000.pcie: use outband MSI support
> [    6.086951] rk-pcie fe190000.pcie: Missing *config* reg space
> [    6.086966] rk-pcie fe190000.pcie: host bridge /pcie@fe190000 ranges:
> [    6.086996] rk-pcie fe190000.pcie:      err 0x00f4000000..0x00f40fffff -> 0x00f4000000
> [    6.087018] rk-pcie fe190000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
> [    6.087040] rk-pcie fe190000.pcie:      MEM 0x00f4200000..0x00f4ffffff -> 0x00f4200000
> [    6.087058] rk-pcie fe190000.pcie:      MEM 0x0a00000000..0x0a3fffffff -> 0x0a00000000
> [    6.087088] rk-pcie fe190000.pcie: Missing *config* reg space
> [    6.087118] rk-pcie fe190000.pcie: invalid resource
> [    6.293620] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x2
> [    6.319156] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x2
> [    6.345823] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x2
> [    6.372490] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x2
> [    6.399152] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x2
> [    6.425824] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x2
> [    6.452485] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x1
> [    6.479154] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x0
> [    6.505824] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x2
> [    6.532487] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x2
> [    8.992510] rk-pcie fe190000.pcie: PCIe Link Fail
> [    8.992526] rk-pcie fe190000.pcie: failed to initialize host
> ```
> 
> This happens with or without the NVMe SSD plugged in.
> 
> Here is the full dmesg log: https://pastebin.com/4UZ03yAP
> 
> Please tell me if this is a software or hardware issue and how to fix this.
> 
> Thanks.
> 
> Also posted on OrangePI forums: http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=144107&extra=

Hey, I'm having the same problem but my ssd is SATA

orangepi-xunlong, 2023-01-05:
> Hey, I'm having the same problem but my ssd is SATA

dts is set to nvme ssd by default, and if using sata ssd please add **overlays=ssd-sata** to **/boot/orangepiEnv.txt** (need reboot).

orangepi@orangepi5:~$ cat /boot/orangepiEnv.txt
verbosity=1
bootlogo=false
overlay_prefix=rk3588
**overlays=ssd-sata**
fdtfile=rockchip/rk3588s-orangepi-5.dtb

closed 2023-01-05 by orangepi-xunlong

fenrirjk, 2023-01-05:
> 
> dts is set to nvme ssd by default, and if using sata ssd please add **overlays=ssd-sata** to **/boot/orangepiEnv.txt** (need reboot).
> 
> orangepi@orangepi5:~$ cat /boot/orangepiEnv.txt verbosity=1 bootlogo=false overlay_prefix=rk3588 **overlays=ssd-sata** fdtfile=rockchip/rk3588s-orangepi-5.dtb

My bad, my disk was actually a SATA (not NVMe) and seems to be correctly recognized:

```
orangepi@orangepi5:~$ dmesg | grep sda
[    6.641487] sd 0:0:0:0: [sda] 234441648 512-byte logical blocks: (120 GB/112 GiB)
[    6.641516] sd 0:0:0:0: [sda] Write Protect is off
[    6.641521] sd 0:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    6.641564] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    6.643254] sd 0:0:0:0: [sda] Attached SCSI disk
```

My question now is:
The following document describes the procedure to boot from a M.2 PCIe SSD. Is it possible to boot from a M.2 SATA SSD ?
https://drive.google.com/file/d/1D0wW6wm7t2Kw9GXyG0tdWDfYp_DQDdeS/view

Thanks.

orangepi-xunlong, 2023-01-05:
1. Burn spi flash

```
cd /usr/share/orangepi5/
sudo dd if=rkspi_loader_sata.img of=/dev/mtdblock0
```

2. Burn linux image to sata ssd

```
sudo dd bs=1M if=Orangepi5_x.x.x_debian_bullseye_desktop_xfce_linux5.10.110.img of=/dev/sda status=progress
sudo sync
```

3. Add overlays=sdd-sata to orangepiEnv.txt of sata ssd (**sudo mount /dev/sda1 /mnt/**   mount the sata ssd /boot partition to add configuration)

fenrirjk, 2023-01-05:
Thanks @orangepi-xunlong ,

I managed to boot from my SSD.

danielpinto8zz6, 2023-01-07:
Hi @orangepi-xunlong can you explain me why another uboot is needed for sata? is a different build? what changes?

orangepi-xunlong, 2023-01-09:
Because pcie and sata cannot be enabled at the same time.

```
test@test:~/orangepi-build$ cat external/config/boards/orangepi5.conf
# Rockchip RK3588s SoC
BOARD_NAME="Orange Pi 5"
BOARDFAMILY="rockchip-rk3588"
BOOTCONFIG="orangepi_5_defconfig"                   # for pcie
#BOOTCONFIG="orangepi_5_sata_defconfig"             # for sata
```

danielpinto8zz6, 2023-01-09:
@orangepi-xunlong thank you 👍

Satyani, 2023-09-26:
I found another solution under Armbian 23.11.0-trunk.147 Lunar with Linux 5.10.160-legacy-rk35xx
for my [5-port SATA contoller](https://aliexpress.ru/item/1005005721252357.html) to work:

`apt install armbian-config` 
Then in `System -> Hardware` I enabled `orangepi-5-sata`

After reboot there is
`overlays=orangepi-5-sata`
instead of
`overlays=ssd-sata`
in  /boot/armbianEnv.txt

and also
/boot/System.map-5.10.160-legacy-rk35xx
/boot/config-5.10.160-legacy-rk35xx
new files appeared.

The "*config* reg space" is no longer "Missing".

Good luck!

fabio-s-franco, 2024-05-21:
Just wanted to add that none of these helped for me.

I have a SATA HD too, the overlays (as listed in `/boot/dtb-6.1.43-rockchip-rk3588/rockchip/overlay` on debian) I needed to enable were two:
`ssd-sata0 ssd-sata2` (separated by spaces)
I also needed to set on board environment file:
`rootdev=/dev/sda2`

Which is where the OS was installed. Without these the link would fail.

MarsThunder, 2024-07-02:
I too was spinning my wheels with an OrangePi3B.  Couldn't figure out how to show NVME in /dev.  For all those scratching their heads, it boils down to this (at least with 3B):  You cannot see MMC (SD Card) and NVME at the same time.  I had a one-notch NVME too.  It does work though!  I had Ubuntu 24.04 server, ran 'sudo nand-sata-install' with option 7, using SDcard, copied same .img with Windows and USB NVME enclosure to the NVME.  Removed SDcard from PI, installed NVME to back of PI, and turned on power.  Booted up instantly and much faster than SDcard. 1TB may be overkill for a PI, but now there's plenty of room.