orangepi-xunlong/orangepi-build #55: spi_busnum_to_master is not exported

alexgubanow, 2022-12-28:
i see no symbol spi_busnum_to_master being exported as well <linux/spi/spi.h> does not have :
`extern struct spi_master *spi_busnum_to_master(u16 busnum);`
If i manually add that declaration, build fails a bit later with 

```
  MODPOST /home/orangepi/driver/Module.symvers
ERROR: modpost: "spi_busnum_to_master" [/home/orangepi/driver/spissd1306.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:134: /home/orangepi/driver/Module.symvers] Error 1
make[2]: *** Deleting file '/home/orangepi/driver/Module.symvers'
make[1]: *** [Makefile:1763: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.16.17-sun50iw9'
make: *** [Makefile:8: all] Error 2
```

the code im trying to build works on other platforms, in fact it is a part of tutorial:
https://github.com/Embetronicx/Tutorials/tree/master/Linux/Device_Driver/SPI_Driver/SPI-driver-example-protocol-driver

leeboby, 2023-01-07:
This driver is used for Linux 5.10.27,  you need to update the driver code to use it in Linux 5.16,  instead of looking for problems with the kernel.

alexgubanow, 2023-01-07:
don't you see the problem ? :)
spissd1306 driver was an example of issue, while the root cause is because spi_busnum_to_master, which is part of kernel, is not exported in 5.16. That is "problems with kernel". If spi_busnum_to_master is not supported in 5.16 it should be wiped from sources than, to avoid misleading.

leeboby, 2023-01-07:
https://github.com/torvalds/linux/commit/bdc7ca008e1f5539e891187032cb2cbbc3decb5e

Since v5.16-rc1,  spi_busnum_to_master() has been removed from the Linux source code, so you should update the driver to adapt to the new kernel source code.

If you still think it's the kernel problem, you can add the code back. It's your freedom.

alexgubanow, 2023-01-07:
https://github.com/orangepi-xunlong/OrangePiH6_kernel/blob/dbcd5dc37b57ea78ffc52fd43e24d6f0af73bc77/drivers/spi/spi.c#L1275
https://github.com/orangepi-xunlong/OrangePiH6_kernel/blob/dbcd5dc37b57ea78ffc52fd43e24d6f0af73bc77/include/linux/spi/spi.h#L421

why this is still here than?

leeboby, 2023-01-07:
This is linux3.10.     

3.10 < 5.16.

closed 2023-01-09 by orangepi-xunlong

lijzijie, 2024-06-07:
Hi @leeboby :
As you said that "spi_busnum_to_master() has been removed from the Linux source cod", is there any other function that can be used as an alternative? or how to get a spi_master/spi_controller when use DTS? Here is my kernel 5.15 code, How should I modify it to use the newer version of the kernel?

`int plcd_spi_driver_add(void)
{
	struct spi_controller *ctlr;
	struct spi_device *spi_device;
	int ret;

	if (!plcd_drv->pcfg->spi_cfg.spi_info) {
		printk("%s: spi_info is null\n", __func__);
		return -1;
	}

	ctlr = spi_busnum_to_master(plcd_drv->pcfg->spi_cfg.spi_info->bus_num);
	if (!ctlr) {
		printk("get busnum failed\n");
		return -1;
	}

	spi_device = spi_new_device(ctlr, plcd_drv->pcfg->spi_cfg.spi_info);
	if (!spi_device) {
		printk("get spi_device failed\n");
		return -1;
	}
	plcd_drv->pcfg->spi_cfg.spi_dev = spi_device;

	ret = spi_register_driver(&plcd_spi_dev_driver);
	if (ret) {
		printk("%s failed\n", __func__);
		return -1;
	}
	if (!plcd_drv->pcfg->spi_cfg.spi_dev) {
		printk("%s failed\n", __func__);
		return -1;
	}

	printk("%s ok\n", __func__);
	return 0;
}`

CeSiumUA, 2024-06-09:
@lijzijie I'm also stuck with this problem. Did you find any solution to it?
`spi_busnum_to_master` is just not present in recent kernel versions

lijzijie, 2024-06-24:
@CeSiumUA My solution is to use DTS to add a spi device instead of using spi_new_device() to add a new spi device. I think you can find a solution with ChatGPT.