orangepi-xunlong/orangepi-build #56: orangepi 4b lts can't boot from emmc

yonoka, 2022-12-30:
orangepi 4b lts 意外停电后，重新上电，出现无法启动问题

看上去像错误的使用/boot/dtb/rockchip/rk3399-orangepi-4.dtb
应该使用的是rk3399-orangepi-4-lts.dtb

分析发现读取orangepiEnv.txt的异常，
`
4096 bytes read in 16 ms (250 KiB/s)
`
实际应该读取209字节。

请问这个问题怎么解决。

完整启动日志
---
`
DDR Version 1.24 20191016
In
channel 0
CS = 0
MR0=0x18
MR4=0x1
MR5=0x1
MR8=0x10
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 1
CS = 0
MR0=0x18
MR4=0x1
MR5=0x1
MR8=0x10
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 0 training pass!
channel 1 training pass!
change freq to 416MHz 0,1
Channel 0: LPDDR4,416MHz
Bus Width=32 Col=10 Bank=8 Row=16 CS=1 Die Bus-Width=16 Size=2048MB
Channel 1: LPDDR4,416MHz
Bus Width=32 Col=10 Bank=8 Row=16 CS=1 Die Bus-Width=16 Size=2048MB
256B stride
channel 0
CS = 0
MR0=0x18
MR4=0x1
MR5=0x1
MR8=0x10
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 1
CS = 0
MR0=0x18
MR4=0x1
MR5=0x1
MR8=0x10
MR12=0x72
MR14=0x72
MR18=0x0
MR19=0x0
MR24=0x8
MR25=0x0
channel 0 training pass!
channel 1 training pass!
channel 0, cs 0, advanced training done
channel 1, cs 0, advanced training done
change freq to 856MHz 1,0
ch 0 ddrconfig = 0x101, ddrsize = 0x40
ch 1 ddrconfig = 0x101, ddrsize = 0x40
pmugrf_os_reg[2] = 0x32C1F2C1, stride = 0xD
ddr_set_rate to 328MHZ
ddr_set_rate to 666MHZ
ddr_set_rate to 928MHZ
channel 0, cs 0, advanced training done
channel 1, cs 0, advanced training done
ddr_set_rate to 416MHZ, ctl_index 0
ddr_set_rate to 856MHZ, ctl_index 1
support 416 856 328 666 928 MHz, current 856MHz
OUT
Boot1: 2019-03-14, version: 1.19
CPUId = 0x0
ChipType = 0x10, 255
SdmmcInit=2 0
BootCapSize=100000
UserCapSize=14910MB
FwPartOffset=2000 , 100000
mmc0:cmd8,20
mmc0:cmd5,20
mmc0:cmd55,20
mmc0:cmd1,20
mmc0:cmd8,20
mmc0:cmd5,20
mmc0:cmd55,20
mmc0:cmd1,20
mmc0:cmd8,20
mmc0:cmd5,20
mmc0:cmd55,20
mmc0:cmd1,20
SdmmcInit=0 1
StorageInit ok = 67833
SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit read PBA: 0x1404
SecureInit read PBA: 0x1804
SecureInit read PBA: 0x1c04
SecureInit ret = 0, SecureMode = 0
atags_set_bootdev: ret:(0)
GPT 0x3380ec0 signature is wrong
recovery gpt...
GPT 0x3380ec0 signature is wrong
recovery gpt fail!
LoadTrust Addr:0x4000
No find bl30.bin
No find bl32.bin
Load uboot, ReadLba = 2000
Load OK, addr=0x200000, size=0xb5130
RunBL31 0x40000
NOTICE:  BL31: v1.3(debug):42583b6
NOTICE:  BL31: Built : 07:55:13, Oct 15 2019
NOTICE:  BL31: Rockchip release version: v1.1
INFO:    GICv3 with legacy support detected. ARM GICV3 driver initialized in EL3
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 0
INFO:    plat_rockchip_pmu_init(1190): pd status 3e
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2020.10-orangepi (Mar 07 2022 - 18:56:22 +0800)

Model: OrangePi 4 AI board
DRAM:  3.9 GiB
PMIC:  RK808
MMC:   mmc@fe320000: 1, sdhci@fe330000: 0
Loading Environment from MMC... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Model: OrangePi 4 AI board
Net:   eth0: ethernet@fe300000
Hit any key to stop autoboot:  0
Card did not respond to voltage select!
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
3341 bytes read in 18 ms (180.7 KiB/s)
Executing script at 00500000

Boot script loaded from mmc 0
4096 bytes read in 16 ms (250 KiB/s)
6501981 bytes read in 637 ms (9.7 MiB/s)
10810271 bytes read in 1048 ms (9.8 MiB/s)
Failed to load '/boot/dtb/rockchip/rk3399-orangepi-4.dtb'
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
No FDT memory address configured. Please configure
the FDT address via "fdt addr <address>" command.
Aborting!
Failed to load '/boot/dtb/rockchip/overlay/-fixup.scr'
   Uncompressing Kernel Image
Moving Image from 0x2080000 to 0x2200000, end=40e0000
Loading init Ramdisk from Legacy Image at 06000000 ...

   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    6501917 Bytes = 6.2 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
ERROR: Did not find a cmdline Flattened Device Tree
Could not find a valid device tree
SCRIPT FAILED: continuing...
starting USB...
Bus usb@fe380000: USB EHCI 1.00
Bus usb@fe3c0000: USB EHCI 1.00
scanning bus usb@fe380000 for devices... 1 USB Device(s) found
scanning bus usb@fe3c0000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found

Device 0: unknown device
ethernet@fe300000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
`

13037639065, 2022-12-30:
需要对emmc进行修复。系统才能从EMMC启动。
用SD启动系统。使用下面方式可以修复MMC
1. `fsck /dev/mmcb1k2p1`
2. 挂载分区，卸载分区。

修复MMC后，拔掉SD卡，系统能够正常从MMC启动。

orangepi-xunlong, 2023-01-03:
> orangepi 4b lts 意外停电后，重新上电，出现无法启动问题
> 
> 看上去像错误的使用/boot/dtb/rockchip/rk3399-orangepi-4.dtb 应该使用的是rk3399-orangepi-4-lts.dtb
> 
> 分析发现读取orangepiEnv.txt的异常， `4096 bytes read in 16 ms (250 KiB/s)` 实际应该读取209字节。
> 
> 请问这个问题怎么解决。
> ## 完整启动日志
> 
> ` DDR Version 1.24 20191016 In channel 0 CS = 0 MR0=0x18 MR4=0x1 MR5=0x1 MR8=0x10 MR12=0x72 MR14=0x72 MR18=0x0 MR19=0x0 MR24=0x8 MR25=0x0 channel 1 CS = 0 MR0=0x18 MR4=0x1 MR5=0x1 MR8=0x10 MR12=0x72 MR14=0x72 MR18=0x0 MR19=0x0 MR24=0x8 MR25=0x0 channel 0 training pass! channel 1 training pass! change freq to 416MHz 0,1 Channel 0: LPDDR4,416MHz Bus Width=32 Col=10 Bank=8 Row=16 CS=1 Die Bus-Width=16 Size=2048MB Channel 1: LPDDR4,416MHz Bus Width=32 Col=10 Bank=8 Row=16 CS=1 Die Bus-Width=16 Size=2048MB 256B stride channel 0 CS = 0 MR0=0x18 MR4=0x1 MR5=0x1 MR8=0x10 MR12=0x72 MR14=0x72 MR18=0x0 MR19=0x0 MR24=0x8 MR25=0x0 channel 1 CS = 0 MR0=0x18 MR4=0x1 MR5=0x1 MR8=0x10 MR12=0x72 MR14=0x72 MR18=0x0 MR19=0x0 MR24=0x8 MR25=0x0 channel 0 training pass! channel 1 training pass! channel 0, cs 0, advanced training done channel 1, cs 0, advanced training done change freq to 856MHz 1,0 ch 0 ddrconfig = 0x101, ddrsize = 0x40 ch 1 ddrconfig = 0x101, ddrsize = 0x40 pmugrf_os_reg[2] = 0x32C1F2C1, stride = 0xD ddr_set_rate to 328MHZ ddr_set_rate to 666MHZ ddr_set_rate to 928MHZ channel 0, cs 0, advanced training done channel 1, cs 0, advanced training done ddr_set_rate to 416MHZ, ctl_index 0 ddr_set_rate to 856MHZ, ctl_index 1 support 416 856 328 666 928 MHz, current 856MHz OUT Boot1: 2019-03-14, version: 1.19 CPUId = 0x0 ChipType = 0x10, 255 SdmmcInit=2 0 BootCapSize=100000 UserCapSize=14910MB FwPartOffset=2000 , 100000 mmc0:cmd8,20 mmc0:cmd5,20 mmc0:cmd55,20 mmc0:cmd1,20 mmc0:cmd8,20 mmc0:cmd5,20 mmc0:cmd55,20 mmc0:cmd1,20 mmc0:cmd8,20 mmc0:cmd5,20 mmc0:cmd55,20 mmc0:cmd1,20 SdmmcInit=0 1 StorageInit ok = 67833 SecureMode = 0 SecureInit read PBA: 0x4 SecureInit read PBA: 0x404 SecureInit read PBA: 0x804 SecureInit read PBA: 0xc04 SecureInit read PBA: 0x1004 SecureInit read PBA: 0x1404 SecureInit read PBA: 0x1804 SecureInit read PBA: 0x1c04 SecureInit ret = 0, SecureMode = 0 atags_set_bootdev: ret:(0) GPT 0x3380ec0 signature is wrong recovery gpt... GPT 0x3380ec0 signature is wrong recovery gpt fail! LoadTrust Addr:0x4000 No find bl30.bin No find bl32.bin Load uboot, ReadLba = 2000 Load OK, addr=0x200000, size=0xb5130 RunBL31 0x40000 �NOTICE: BL31: v1.3(debug):42583b6 NOTICE: BL31: Built : 07:55:13, Oct 15 2019 NOTICE: BL31: Rockchip release version: v1.1 INFO: GICv3 with legacy support detected. ARM GICV3 driver initialized in EL3 INFO: Using opteed sec cpu_context! INFO: boot cpu mask: 0 INFO: plat_rockchip_pmu_init(1190): pd status 3e INFO: BL31: Initializing runtime services WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK ERROR: Error initializing runtime service opteed_fast INFO: BL31: Preparing for EL3 exit to normal world INFO: Entry point address = 0x200000 INFO: SPSR = 0x3c9
> 
> U-Boot 2020.10-orangepi (Mar 07 2022 - 18:56:22 +0800)
> 
> Model: OrangePi 4 AI board DRAM: 3.9 GiB PMIC: RK808 MMC: mmc@fe320000: 1, sdhci@fe330000: 0 Loading Environment from MMC... *** Warning - bad CRC, using default environment
> 
> In: serial Out: serial Err: serial Model: OrangePi 4 AI board Net: eth0: ethernet@fe300000 Hit any key to stop autoboot: 0 Card did not respond to voltage select! switch to partitions #0, OK mmc0(part 0) is current device Scanning mmc 0:1... Found U-Boot script /boot/boot.scr 3341 bytes read in 18 ms (180.7 KiB/s) Executing script at 00500000
> 
> Boot script loaded from mmc 0 4096 bytes read in 16 ms (250 KiB/s) 6501981 bytes read in 637 ms (9.7 MiB/s) 10810271 bytes read in 1048 ms (9.8 MiB/s) Failed to load '/boot/dtb/rockchip/rk3399-orangepi-4.dtb' libfdt fdt_check_header(): FDT_ERR_BADMAGIC No FDT memory address configured. Please configure the FDT address via "fdt addr
> " command.
> Aborting!
> Failed to load '/boot/dtb/rockchip/overlay/-fixup.scr'
> Uncompressing Kernel Image
> Moving Image from 0x2080000 to 0x2200000, end=40e0000
> Loading init Ramdisk from Legacy Image at 06000000 ...
> 
> Image Name: uInitrd Image Type: AArch64 Linux RAMDisk Image (gzip compressed) Data Size: 6501917 Bytes = 6.2 MiB Load Address: 00000000 Entry Point: 00000000 Verifying Checksum ... OK ERROR: Did not find a cmdline Flattened Device Tree Could not find a valid device tree SCRIPT FAILED: continuing... starting USB... Bus usb@fe380000: USB EHCI 1.00 Bus usb@fe3c0000: USB EHCI 1.00 scanning bus usb@fe380000 for devices... 1 USB Device(s) found scanning bus usb@fe3c0000 for devices... 1 USB Device(s) found scanning usb for storage devices... 0 Storage Device(s) found
> 
> Device 0: unknown device ethernet@fe300000 Waiting for PHY auto negotiation to complete......... TIMEOUT ! `

这种现象直接重烧系统即可。

orangepi-xunlong, 2023-01-03:
> 我也遇到这个问题。 并且发现以下的现象：
> 
> 需要对emmc进行修复。系统才能从EMMC启动。 用SD启动系统。使用下面方式可以修复MMC
> 
>     1. `fsck /dev/mmcb1k2p1`
> 
>     2. 挂载分区，卸载分区。
> 
> 
> 修复MMC后，拔掉SD卡，系统能够正常从MMC启动。
> 
> 如何能够自动修复磁盘并正常的进入系统呢？ 否则掉电后，就需要救机器

可以研究下overlayroot，Ubuntu系统中直接安装配置使用应该没啥问题，具体可以百度找下资料。

closed 2023-01-03 by orangepi-xunlong

kingsimba, 2023-01-05:
经过排查，发现 `/boot/orangepiEnv.txt` 这个文件每次启动都会被修改。所以意外掉电的话，这个文件会有大概率坏。
能否让这个文件不要修改，就会大大降低出问题的概率。

orangepi-xunlong, 2023-01-05:
把 /usr/lib/orangepi/orangepi-hardware-optimization 中的 add_usb_storage_quirks & 屏蔽掉试下。

kingsimba, 2023-01-05:
> 把 /usr/lib/orangepi/orangepi-hardware-optimization 中的 add_usb_storage_quirks & 屏蔽掉试下。

可行！ 屏蔽 `add_usb_storage_quirks &` 后， /boot/orangepiEnv.txt 不再改变。问题大概率解决了，我再排查下有没有别的服务会修改启动时的关键文件。