orangepi-xunlong/orangepi-build #57: Failed to retrieve jammy release on non-China mirror

beriberikix, 2023-01-04:
Uncommented the China download mirror in `config-default.conf` and when I went to build a Jammy server image it failed.

```
[ o.k. ] Installing base system [ Stage 1/2 ]
I: Retrieving InRelease 
I: Retrieving Release 
E: Failed getting release file https://ports.ubuntu.com/dists/jammy/Release
[ error ] ERROR in function create_rootfs_cache [ build.sh:305 -> scripts/main.sh:603 -> scripts/main.sh:572 -> scripts/debootstrap.sh:50 -> scripts/debootstrap.sh:162 -> scripts/general.sh:0 ]
[ error ] Debootstrap base system for legacy orangepi5 jammy   yes first stage failed 
[ o.k. ] Process terminated 
[ error ] unmount_on_exit() called! [ build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:0 ]
[ o.k. ] Unmounting [ /home/jberi/orangepi-build/.tmp/rootfs-a5df2e4e-9e48-40a2-beb3-3d1e26833c38/ ]
[ error ] ERROR in function unmount_on_exit [ build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:90 -> scripts/general.sh:0 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

orangepi-xunlong, 2023-01-05:
```
--- a/scripts/debootstrap.sh
+++ b/scripts/debootstrap.sh
@@ -137,7 +137,7 @@ create_rootfs_cache()
                        local apt_extra="-o Acquire::http::Proxy=\"http://${APT_PROXY_ADDR:-localhost:3142}\""
                        local apt_mirror="http://${APT_PROXY_ADDR:-localhost:3142}/$APT_MIRROR"
                else
-                       local apt_mirror="https://$APT_MIRROR"
+                       local apt_mirror="http://$APT_MIRROR"
```

Please try this.

beriberikix, 2023-01-05:
That solved the issue! It also fixes #58, which was probably more of a result of the mirrors not working.

closed 2023-01-05 by beriberikix