orangepi-xunlong/orangepi-build #58: Build fails on rtl88x2cs driver for Orange Pi 5

beriberikix, 2023-01-04:
Hello! I just tried building a minimal server image with standard configurations on my ubuntu desktop and the build failed when getting to the rtl88x2cs. I had to disable the "Rockchip Wireless LAN" driver to get past this error. I'm on the `next` branch.

```
make[4]: *** [scripts/Makefile.build:273: drivers/net/wireless/rtl88x2cs/hal/hal_hci/hal_sdio_coex.o] Error 1
make[3]: *** [scripts/Makefile.build:516: drivers/net/wireless/rtl88x2cs] Error 2
fixdep: error opening file: drivers/net/wireless/rtl88x2bu/hal/phydm/.phydm_psd.o.d: No such file or directory
make[4]: *** [scripts/Makefile.build:273: drivers/net/wireless/rtl88x2bu/hal/phydm/phydm_psd.o] Error 2
make[3]: *** [scripts/Makefile.build:516: drivers/net/wireless/rtl88x2bu] Error 2
make[5]: *** [scripts/Makefile.build:450: drivers/net/wireless/rockchip_wlan/rtl8852be/8852be.o] Error 1
make[4]: *** [scripts/Makefile.build:516: drivers/net/wireless/rockchip_wlan/rtl8852be] Error 2
make[3]: *** [scripts/Makefile.build:516: drivers/net/wireless/rockchip_wlan] Error 2
make[2]: *** [scripts/Makefile.build:516: drivers/net/wireless] Error 2
make[1]: *** [scripts/Makefile.build:516: drivers/net] Error 2
make: *** [Makefile:1920: drivers] Error 2
[ error ] ERROR in function compile_kernel [ build.sh:305 -> scripts/main.sh:603 -> scripts/main.sh:503 -> scripts/compilation.sh:499 -> scripts/general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated 
```

The image still isn't building because of https://github.com/orangepi-xunlong/orangepi-build/issues/57 but that's a separate issue.

orangepi-xunlong, 2023-01-05:
I cannot reproduce this error, can you provide more information？

```
test@test:~/orangepi-build$ ls kernel/orange-pi-5.10-rk3588/drivers/net/wireless/rtl88x2cs/
88x2cs.ko   88x2cs.mod.c  88x2cs.o  hal        ifcfg-wlan0  Kconfig   modules.order  platform     runwpa
88x2cs.mod  88x2cs.mod.o  core      halmac.mk  include      Makefile  os_dep         rtl8822c.mk  wlan0dhcp
```

beriberikix, 2023-01-05:
Fixed via #57

closed 2023-01-05 by beriberikix