orangepi-xunlong/orangepi-build #59: build kernel failed

ignorantimt, 2023-01-05:
OS: `Ubuntu 22.04.1 LTS`
code: `git clone https://github.com/orangepi-xunlong/orangepi-build.git -b next`
user: `root`

After running `./build.sh', I first chose u-boot, and there are two issues:
1. When compiling, it read: `Compiler version [ aarch64-linux-gnu-gcc 11 ]`
2. After that, `orangepi-build/toolchains` is empty

Then, I run `./build.sh`, this time I chose kernel, new issue occurred:
```
env: ‘aarch64-none-linux-gnu-gcc’: No such file or directory
[ o.k. ] Compiler version [ aarch64-none-linux-gnu-gcc  ]
[ o.k. ] Using kernel config file [ /root/orangepi-build/external/config/kernel/linux-rockchip-rk3588-legacy.config ]
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/conf
./scripts/gcc-version.sh: 20: printf: ccache:: expected numeric value
./scripts/gcc-version.sh: 20: printf: error:: expected numeric value
./scripts/gcc-version.sh: 20: printf: Could: expected numeric value
./scripts/gcc-version.sh: 20: printf: not: expected numeric value
./scripts/gcc-version.sh: 20: printf: find: expected numeric value
./scripts/gcc-version.sh: 20: printf: compiler: expected numeric value
./scripts/gcc-version.sh: 20: printf: in: expected numeric value
./scripts/gcc-version.sh: 20: printf: PATH: expected numeric value
./scripts/gcc-version.sh: 20: printf: ccache:: expected numeric value
./scripts/gcc-version.sh: 20: printf: error:: expected numeric value
./scripts/gcc-version.sh: 20: printf: Could: expected numeric value
./scripts/gcc-version.sh: 20: printf: not: expected numeric value
./scripts/gcc-version.sh: 20: printf: find: expected numeric value
./scripts/gcc-version.sh: 20: printf: compiler: expected numeric value
./scripts/gcc-version.sh: 20: printf: in: expected numeric value
./scripts/gcc-version.sh: 20: printf: PATH: expected numeric value
./scripts/gcc-version.sh: 20: printf: ccache:: expected numeric value
./scripts/gcc-version.sh: 20: printf: error:: expected numeric value
./scripts/gcc-version.sh: 20: printf: Could: expected numeric value
./scripts/gcc-version.sh: 20: printf: not: expected numeric value
./scripts/gcc-version.sh: 20: printf: find: expected numeric value
./scripts/gcc-version.sh: 20: printf: compiler: expected numeric value
./scripts/gcc-version.sh: 20: printf: in: expected numeric value
./scripts/gcc-version.sh: 20: printf: PATH: expected numeric value
ccache: error: Could not find compiler "aarch64-none-linux-gnu-ld" in PATH
init/Kconfig:39: syntax error
init/Kconfig:38: invalid statement
ccache: error: Could not find compiler "aarch64-none-linux-gnu-ld" in PATH
make[1]: *** [scripts/kconfig/Makefile:71: oldconfig] Error 1
make: *** [Makefile:627: oldconfig] Error 2
  UPD     scripts/kconfig/mconf-cfg
  HOSTCC  scripts/kconfig/mconf.o
  HOSTCC  scripts/kconfig/lxdialog/checklist.o
  HOSTCC  scripts/kconfig/lxdialog/inputbox.o
  HOSTCC  scripts/kconfig/lxdialog/menubox.o
  HOSTCC  scripts/kconfig/lxdialog/textbox.o
  HOSTCC  scripts/kconfig/lxdialog/util.o
  HOSTCC  scripts/kconfig/lxdialog/yesno.o
  HOSTLD  scripts/kconfig/mconf
./scripts/gcc-version.sh: 20: printf: ccache:: expected numeric value
./scripts/gcc-version.sh: 20: printf: error:: expected numeric value
./scripts/gcc-version.sh: 20: printf: Could: expected numeric value
./scripts/gcc-version.sh: 20: printf: not: expected numeric value
./scripts/gcc-version.sh: 20: printf: find: expected numeric value
./scripts/gcc-version.sh: 20: printf: compiler: expected numeric value
./scripts/gcc-version.sh: 20: printf: in: expected numeric value
./scripts/gcc-version.sh: 20: printf: PATH: expected numeric value
./scripts/gcc-version.sh: 20: printf: ccache:: expected numeric value
./scripts/gcc-version.sh: 20: printf: error:: expected numeric value
./scripts/gcc-version.sh: 20: printf: Could: expected numeric value
./scripts/gcc-version.sh: 20: printf: not: expected numeric value
./scripts/gcc-version.sh: 20: printf: find: expected numeric value
./scripts/gcc-version.sh: 20: printf: compiler: expected numeric value
./scripts/gcc-version.sh: 20: printf: in: expected numeric value
./scripts/gcc-version.sh: 20: printf: PATH: expected numeric value
./scripts/gcc-version.sh: 20: printf: ccache:: expected numeric value
./scripts/gcc-version.sh: 20: printf: error:: expected numeric value
./scripts/gcc-version.sh: 20: printf: Could: expected numeric value
./scripts/gcc-version.sh: 20: printf: not: expected numeric value
./scripts/gcc-version.sh: 20: printf: find: expected numeric value
./scripts/gcc-version.sh: 20: printf: compiler: expected numeric value
./scripts/gcc-version.sh: 20: printf: in: expected numeric value
./scripts/gcc-version.sh: 20: printf: PATH: expected numeric value
ccache: error: Could not find compiler "aarch64-none-linux-gnu-ld" in PATH
init/Kconfig:39: syntax error
init/Kconfig:38: invalid statement
ccache: error: Could not find compiler "aarch64-none-linux-gnu-ld" in PATH
make[1]: *** [scripts/kconfig/Makefile:29: menuconfig] Error 1
make: *** [Makefile:627: menuconfig] Error 2
[ error ] ERROR in function compile_kernel [ ./build.sh:305 -> scripts/main.sh:603 -> scripts/main.sh:503 -> scripts/compilation.sh:465 -> scripts/general.sh:0 ]
[ error ] Error kernel menuconfig failed
[ o.k. ] Process terminated
```

These issues may be caused by the same reason, and I think it's that cross-compile-toolchains were not downloaded but I don't know how to fix it.

orangepi-xunlong, 2023-01-05:
**rm orangepi-build/toolchains -r** and try **./build.sh** again.

ignorantimt, 2023-01-06:
> **rm orangepi-build/toolchains -r** and try **./build.sh** again.

Thanks for your reply. I've tried it, but it didn't work.

ignorantimt, 2023-01-06:
By the way, the ubuntu image that is used on my Orange Pi 5 board was downloaded from http://www.orangepi.cn/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-pi-5.html (->[BaiduNetdisk](https://pan.baidu.com/share/init?surl=Qexm8NT7GJ1JTd4jAhxDtQ)).

orangepi-xunlong, 2023-01-06:
Please use X64 Ubuntu22.04.

ignorantimt, 2023-01-06:
> Please use X64 Ubuntu22.04.

It's ok. Thanks a lot.

closed 2023-01-06 by ignorantimt