orangepi-xunlong/orangepi-build #61: orange pi zero 2 no /dev/i2c-*

DavidVorick, 2023-01-07:
I am trying to use the GPIOs on the Orange Pi Zero 2, specifically the i2c pins. When I look at /dev/, I see /dev/i2c-0 (which is for HDMI), but I do not see any other i2c options. 

I read through the manual (https://linux-sunxi.org/images/7/72/OrangePi_Zero2_H616_User_Manual_v1.0.pdf) and set up wiringOP, but got stuck because it still could not find /dev/i2c-3 or /dev/i2c-5.

How do I configure the build so that I can use i2c?

DavidVorick, 2023-01-07:
I am happy to confirm that this solved the issue for me, my i2c is now working as hoped.

Is there some place I should have looked in the documentation?

leeboby, 2023-01-07:
https://drive.google.com/drive/folders/1ToDjWZQptABxfiRwaeYW1WzQILM5iwpb

[image]

[image]

closed 2023-01-08 by DavidVorick