orangepi-xunlong/orangepi-build #67: orangepi 4 TLS wifi station模式下出现问题

13037639065, 2023-02-01:
问题描述
orangepi 4 TLS的wifi在station模式下
出问题前的系统日志
```
Feb  1 14:14:44 23922073002193x NetworkManager[1981]: <info>  [1675232084.8759] policy: set 'AutoXing' (wlan0) as default for IPv4 routing and DNS
Feb  1 14:19:45 23922073002193x wpa_supplicant[2034]: wlan0: Trying to associate with 00:be:d5:70:8e:e2 (SSID='AutoXing' freq=5260 MHz)
Feb  1 14:19:45 23922073002193x wpa_supplicant[2034]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:be:d5:70:8e:e2 status_code=1
Feb  1 14:19:45 23922073002193x NetworkManager[1981]: <info>  [1675232385.2921] device (wlan0): supplicant interface state: completed -> associating
Feb  1 14:19:45 23922073002193x NetworkManager[1981]: <info>  [1675232385.2923] device (p2p-dev-wlan0): supplicant management interface state: completed -> associating
Feb  1 14:19:45 23922073002193x NetworkManager[1981]: <info>  [1675232385.3121] device (wlan0): supplicant interface state: associating -> disconnected
Feb  1 14:19:45 23922073002193x NetworkManager[1981]: <info>  [1675232385.3123] device (p2p-dev-wlan0): supplicant management interface state: associating -> disconnected
Feb  1 14:19:45 23922073002193x NetworkManager[1981]: <info>  [1675232385.4058] device (wlan0): supplicant interface state: disconnected -> scanning
Feb  1 14:19:45 23922073002193x NetworkManager[1981]: <info>  [1675232385.4059] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.0462] agent-manager: agent[df2b65849ec8212a,:1.130/nmcli-connect/0]: agent registered
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.0516] device (wlan0): state change: activated -> deactivating (reason 'new-activation', sys-iface-state: 'managed')
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.0587] device (wlan0): disconnecting for new activation request.
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.0589] audit: op="connection-activate" uuid="bd21adc1-75e6-4968-adc8-ab45e456f4b7" name="AutoXing" pid=1210806 uid=0 result="success"
Feb  1 14:19:46 23922073002193x dbus-daemon[616]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.9' (uid=0 pid=1981 comm="/usr/sbin/NetworkManager --no-daemon ")
Feb  1 14:19:46 23922073002193x systemd[1]: Starting Network Manager Script Dispatcher Service...
Feb  1 14:19:46 23922073002193x dbus-daemon[616]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Feb  1 14:19:46 23922073002193x systemd[1]: Started Network Manager Script Dispatcher Service.
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <warn>  [1675232386.1156] device (wlan0): Deactivation failed: GDBus.Error:fi.w1.wpa_supplicant1.NotConnected: This interface is not connected
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1161] device (wlan0): state change: deactivating -> disconnected (reason 'new-activation', sys-iface-state: 'managed')
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1443] dhcp4 (wlan0): canceled DHCP transaction
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1444] dhcp4 (wlan0): state changed bound -> done
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1644] policy: set 'Wired connection 1' (usb0) as default for IPv4 routing and DNS
Feb  1 14:19:46 23922073002193x wpa_supplicant[2034]: wlan0: Reject scan trigger since one is already pending
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1661] device (wlan0): Activation: starting connection 'AutoXing' (bd21adc1-75e6-4968-adc8-ab45e456f4b7)
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1759] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1795] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1821] device (wlan0): Activation: (wifi) access point 'AutoXing' has security, but secrets are required.
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1823] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.1980] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.2011] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.2036] device (wlan0): Activation: (wifi) connection 'AutoXing' has security, and secrets exist.  No new secrets needed.
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.2040] Config: added 'ssid' value 'AutoXing'
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.2043] Config: added 'scan_ssid' value '1'
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.2044] Config: added 'bgscan' value 'simple:30:-65:300'
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.2056] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256 FT-PSK'
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.2058] Config: added 'auth_alg' value 'OPEN'
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.2059] Config: added 'psk' value '<hidden>'
Feb  1 14:19:46 23922073002193x wpa_supplicant[2034]: wlan0: Trying to associate with 00:be:d5:70:e8:e2 (SSID='AutoXing' freq=5260 MHz)
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.9038] device (wlan0): supplicant interface state: scanning -> associating
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.9039] device (p2p-dev-wlan0): supplicant management interface state: scanning -> associating
Feb  1 14:19:46 23922073002193x wpa_supplicant[2034]: wlan0: CTRL-EVENT-DISCONNECTED bssid=00:be:d5:70:e8:e2 reason=0
Feb  1 14:19:46 23922073002193x start_all[985163]: #033[0m[ INFO]: /camera_node: CameraSetupResponse : 0 0#033[0m
Feb  1 14:19:46 23922073002193x wpa_supplicant[2034]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Feb  1 14:19:46 23922073002193x wpa_supplicant[2034]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=US
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.9548] device (wlan0): supplicant interface state: associating -> disconnected
Feb  1 14:19:46 23922073002193x NetworkManager[1981]: <info>  [1675232386.9550] device (p2p-dev-wlan0): supplicant management interface state: associating -> disconnected
```
kernel日志
```
Feb  1 14:19:47 23922073002193x kernel: [79134.934742] ------------[ cut here ]------------
Feb  1 14:19:47 23922073002193x kernel: [79134.934865] WARNING: CPU: 2 PID: 1207341 at net/wireless/sme.c:757 __cfg80211_connect_result+0x64/0x1bc [cfg80211]
Feb  1 14:19:47 23922073002193x kernel: [79134.934870] Modules linked in: rtc_ds1307 xt_MASQUERADE iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_filter hci_uart asix qmi_wwan rndis_host cdc_ether cdc_wdm option usbnet usb_wwan fusb30x(C) ch341 sprdwl_ng cfg80211 sprdbt_tty ip_tables x_tables autofs4 pwm_bl motorcomm
Feb  1 14:19:47 23922073002193x kernel: [79134.934963] CPU: 2 PID: 1207341 Comm: kworker/u12:1 Tainted: G        WC        5.10.43 #3.0.0
Feb  1 14:19:47 23922073002193x kernel: [79134.934966] Hardware name: OrangePi 4 LTS (DT)
Feb  1 14:19:47 23922073002193x kernel: [79134.935033] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
Feb  1 14:19:47 23922073002193x kernel: [79134.935043] pstate: 40000005 (nZcv daif -PAN -UAO -TCO BTYPE=--)
Feb  1 14:19:47 23922073002193x kernel: [79134.935106] pc : __cfg80211_connect_result+0x64/0x1bc [cfg80211]
Feb  1 14:19:47 23922073002193x kernel: [79134.935169] lr : __cfg80211_connect_result+0x148/0x1bc [cfg80211]
Feb  1 14:19:47 23922073002193x kernel: [79134.935173] sp : ffffffc0195b3cf0
Feb  1 14:19:47 23922073002193x kernel: [79134.935176] x29: ffffffc0195b3cf0 x28: 0000000000000000 
[... cut ...]
Feb  1 14:19:47 23922073002193x kernel: [79134.935560]  cfg80211_event_work+0x28/0x3c [cfg80211]
Feb  1 14:19:47 23922073002193x kernel: [79134.935572]  process_one_work+0x1d8/0x290
Feb  1 14:19:47 23922073002193x kernel: [79134.935578]  worker_thread+0x1e0/0x270
Feb  1 14:19:47 23922073002193x kernel: [79134.935585]  kthread+0xf4/0x104
Feb  1 14:19:47 23922073002193x kernel: [79134.935593]  ret_from_fork+0x10/0x30
Feb  1 14:19:47 23922073002193x kernel: [79134.935599] ---[ end trace 5fd2a6839aa04e62 ]---
```
wifi 蓝牙功能异常

orangepi-xunlong, 2023-02-03:
你好，具体哪个系统，能稳定复现吗，详细的复现步骤是什么？

13037639065, 2023-02-03:
相关问题 
https://github.com/raspberrypi/linux/issues/3318
https://forum.armbian.com/topic/8408-orange-pi-lite-wireless-issue-netwirelesssmec752-__cfg80211_connect_result/
没有稳定复现的步骤。比较随机

13037639065, 2023-02-28:
```
[13682.884046] WCN: marlin power state:5, subsys: [MARLIN_WIFI] power 0
[13682.884059] WCN: marlin power state:101, subsys: [MARLIN_BLUETOOTH] power 0
```
出现该日志蓝牙WIFI功能就异常。请问下，这个代表的含义是什么呢？除了整机重启是否还有其他的方式能够进行自恢复

forose, 2025-01-08:
> 使用系统： Orangepi4-lts_3.0.0_ubuntu_focal_linux5.10.43_cli_standard 相关问题 [raspberrypi/linux#3318](https://github.com/raspberrypi/linux/issues/3318) https://forum.armbian.com/topic/8408-orange-pi-lite-wireless-issue-netwirelesssmec752-__cfg80211_connect_result/ 没有稳定复现的步骤。比较随机 使用香橙派连接WIFI。偶现该问题

我的是60天左右必出现一次