orangepi-xunlong/orangepi-build #71: kernel build failed for orangpi5(rk3588s)

sunnythree, 2023-02-21:
make[2]: *** [scripts/Makefile.build:273: kernel/sched/core.o] Error 1
make[1]: *** [scripts/Makefile.build:516: kernel/sched] Error 2
make: *** [Makefile:1920: kernel] Error 2
[ error ] ERROR in function compile_kernel [ build.sh:305 -> scripts/main.sh:601 -> scripts/main.sh:504 -> scripts/compilation.sh:496 -> scripts/general.sh:0 ]
[ error ] Kernel was not built [ @host ]

It happens lot of times, I can not resovle it, Can anyone give some help?

orangepi-xunlong, 2023-02-21:
1. 先下载Orange Pi 5的产品使用说明书，按照里面第四章orangepi-build的使用说明操作。确保使用的代码分支为next分支，使用的编译环境ubuntu22.04，还需确保编译系统能正常访问github才能下载好需要的源码。

Orange Pi 5的产品使用说明书下载链接如下所示：
http://www.orangepi.cn/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-pi-5.html

2. 如果还有问题，提交详细的使用步骤和完整的log，不要只发部分错误log。

sunnythree, 2023-02-21:
感谢！！！

closed 2023-02-23 by orangepi-xunlong