orangepi-xunlong/orangepi-build #73: Orange Pi Zero 2 - UART5

rbn920, 2023-03-06:
Page 144 of the Orange Pi Zero 2 give instructions on enabling and testing the UART5 pins. The loopback test does not display results as the manual shows. As an additional test I connected the Orange Pi Zero 2 to a Raspberry Pi4 via UART pins and opened minicom on each device. The Orange Pi Zero 2 will receive but not send.

orangepi-xunlong, 2023-03-07:
Which linux image?

rbn920, 2023-03-07:
Orangepizero2_3.0.6_debian_bullseye_server_linux5.16.17

orangepi-xunlong, 2023-03-07:
Please add overlays=uart5 in /boot/orangepiEnv.txt.

orangepi@orangepi:~$ sudo vim /boot/orangepiEnv.txt
overlays=uart5

rbn920, 2023-03-07:
That was done, just as noted in the manual. Testing using gpio as also noted in the manual does yield correct results. Also when connected to a Pi4 I could receive commands but not send them using minicom.

orangepi-xunlong, 2023-03-07:
???  The loopback test does not display results as the manual shows. ???

rbn920, 2023-03-07:
The manual (p. 145) has you wire the tx pin to the rx pin (8 to 10) then run `gpio serial /dev/ttyS5` with results like:
```
Out: 0: -> 0
Out: 1: -> 1
```
but instead I get:
```
Out: 0: 
Out: 1: 
```

orangepi-xunlong, 2023-03-07:
Are the pins connected correctly？

If the settings are all correct, the hardware is broken.

rbn920, 2023-03-07:
Yes, the pins are connected correctly. I guess my board is bad then.

closed 2023-03-07 by orangepi-xunlong