orangepi-xunlong/orangepi-build #75: Touch support

adrianrb, 2023-03-20:
Hi all!
I bought a OPi Zero 2 with Expansion board, and it´s wonderfull!
Small, fast and usefull! Congratulations!

But, I bought [this](https://pt.aliexpress.com/item/1005005132376545.html?spm=a2g0o.order_list.order_list_main.5.164ecaa47uX9eO&gatewayAdapt=glo2bra) display for testing (yes, I know, it´s for RPi...), and worked beautfully (crispy image), except for touch ("usb" touch type).
Is there any way to make it happen? (config, ".so" driver loading, rebuild kernel/driver etc)

Adrian

baiywt, 2025-04-08:
@adrianrb The default system should support USB touchscreens. Could you let us know the model of the touchscreen you're using? We can help check the driver support based on that.

adrianrb, 2025-04-08:
Whoooa! More than two years later...
OK, the display is below.

[7 Inch Portable Display IPS LCD Capacitive Touch Screen 1024x600 Monitor Optional Speakers for Raspberry Pi.pdf](https://github.com/user-attachments/files/19654809/7.Inch.Portable.Display.IPS.LCD.Capacitive.Touch.Screen.1024x600.Monitor.Optional.Speakers.for.Raspberry.Pi.pdf)

baiywt, 2025-04-09:
@adrianrb Could you please connect the touchscreen and run the following command to check if there are any relevant messages?

> dmesg | grep "touch"