orangepi-xunlong/orangepi-build #82: (bug) unable to build / install zfs on orangepi 5

thefiredragon, 2023-04-22:
Dear all and first thanks your work,
Sadly it's impossible at the moment to build zfs dkms modules under kernel 5.10 and hope this could be fixed in future,
Best Regards
Dan

leeboby, 2023-06-06:
1. The latest orangepi5 v1.1.6 ubuntu/debian has solved this problem.
2. The desktop version images has been preinstalled with zfs and can be used directly.

thefiredragon, 2023-06-12:
Sadly it's not currently working.
If I install latest image from here:
https://drive.google.com/drive/folders/1F2uc8v_EQnvsNrevDihwoymOJlFgM-dZ
Kernel Headers:
https://drive.google.com/drive/folders/1SiUkTWQ5X7U07e5GJvLirqtrVRCIxfr5

I'm running into this issue:

`Setting up zfs-dkms (2.0.3-9+deb11u1) ...
Removing old zfs-2.0.3 DKMS files...

------------------------------
Deleting module version: 2.0.3
completely from the DKMS tree.
------------------------------
Done.
Loading new zfs-2.0.3 DKMS files...
Building for 5.10.110-rockchip-rk3588
Building initial module for 5.10.110-rockchip-rk3588
configure: error:
        *** None of the expected "PDE_DATA" interfaces were detected.
        *** This may be because your kernel version is newer than what is
        *** supported, or you are using a patched custom kernel with
        *** incompatible modifications.
        ***
        *** ZFS Version: zfs-2.0.3-9+deb11u1
        *** Compatible Kernels: 3.10 - 5.10

Error! Bad return status for module build on kernel: 5.10.110-rockchip-rk3588 (aarch64)
Consult /var/lib/dkms/zfs/2.0.3/build/make.log for more information.
dpkg: error processing package zfs-dkms (--configure):
 installed zfs-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of zfs-zed:
 zfs-zed depends on zfs-modules | zfs-dkms; however:
  Package zfs-modules is not installed.
  Package zfs-dkms which provides zfs-modules is not configured yet.
  Package zfs-dkms is not configured yet.

dpkg: error processing package zfs-zed (--configure):
 dependency problems - leaving unconfigured
Processing triggers for initramfs-tools (0.140) ...
ln: failed to create hard link '/boot/initrd.img-5.10.110-rockchip-rk3588.dpkg-bak' => '/boot/initrd.img-5.10.110-rockchip-rk3588': Operation not permitted
update-initramfs: Generating /boot/initrd.img-5.10.110-rockchip-rk3588
update-initramfs: Converting to u-boot format
Errors were encountered while processing:
 zfs-dkms
 zfs-zed
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@armnode01:~#`

leeboby, 2023-06-15:
Please use the kernel header file in the /opt directory, not in Google Drive, the latest image /opt has kernel header file.

In addition, ubuntu22.04 and debian11 cannot install zfs using the apt command.

If you want to install zfs yourself, use the deb packages here: 
https://drive.google.com/drive/folders/11S6k25H8dhlrP4nP0miTxqas3viA8FVG

[image]

Again, the latest desktop version of the images can use zfs directly.

joshdick, 2023-10-03:
I'm curious if there's any update on this issue? (I am experiencing the same symptoms as @thefiredragon when attempting to install Ubuntu's first-party ZFS packages.)

I appreciate that you've provided prebuilt packages to work around the issue, but It would be nice to be able to install the official/first-party Ubuntu ZFS packages using the `apt` command for increased confidence in both longer-term stability and security.

closed 2023-11-22 by orangepi-xunlong