orangepi-xunlong/orangepi-build #88: Orange Pi 5 m2 sata-ssd random eth0 MAC-address every boot

soda-pop-ice-cream, 2023-06-07:
Image: Orangepi5_1.1.6_ubuntu_jammy_server_linux5.10.110.img

When using SD-card, I can reboot board and eth0 MAC-address stays same every boot, but when using m2 sata ssd eth0 MAC-address changing every boot, how can i make it work like with SD-card?

---

What i tried so far is manually set MAC-address in /etc/network/interfaces like this:

```
root@orangepi5:~# cat /etc/network/interfaces
source /etc/network/interfaces.d/*
# Network is managed by Network manager
auto lo
iface lo inet loopback

auto eth0
iface eth0 inet dhcp
  hwaddress ca:15:a1:5f:e0:4c
```

This resulted 2 IP-addresses, one that i wanted and one more random:

```
root@orangepi5:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether ca:15:a1:5f:e0:4c brd ff:ff:ff:ff:ff:ff permaddr 0e:75:ee:62:f5:34
    inet 192.168.1.205/24 brd 192.168.1.255 scope global dynamic noprefixroute eth0
       valid_lft 42830sec preferred_lft 42830sec
    inet 192.168.1.206/24 brd 192.168.1.255 scope global secondary dynamic eth0
       valid_lft 42841sec preferred_lft 42841sec
    inet6 fd25:810f:3273::9ae/128 scope global dynamic noprefixroute 
       valid_lft 42831sec preferred_lft 42831sec
    inet6 fd25:810f:3273:0:31e1:3a5c:5be0:2705/64 scope global temporary dynamic 
       valid_lft 604431sec preferred_lft 85591sec
    inet6 fd25:810f:3273:0:90ce:5e11:7f79:fa64/64 scope global mngtmpaddr noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::a846:e94a:3885:b3c3/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
```

soda-pop-ice-cream, 2023-06-08:
Update: **1.1.4** images have no such issue

orangepi-xunlong, 2023-11-22:
Already resolved.

closed 2023-11-22 by orangepi-xunlong