orangepi-xunlong/orangepi-build #90: Orange Pi R1 Plus LTS Kernel Panic

AlexanderSlaa, 2023-07-01:
Im trying to get my Orange Pi R1 Plus LTS running again.
It was working for a few months now i rebooted it wont boot up again.

```
DDR version 1.16 20190528
ID:0x805 N
In
DDR4
333MHz
Bus Width=32 Col=10 Bank=4 Bank Group=2 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
ddrconfig:14
OUT
Boot1 Release Time: May 13 2019 17:34:36, version: 2.50
ChipType = 0x11, 248
mmc2:cmd1,20
emmc reinit
mmc2:cmd1,20
emmc reinit
mmc2:cmd1,20
SdmmcInit=2 1
mmc0:cmd5,20
SdmmcInit=0 0
BootCapSize=0
UserCapSize=7527MB
FwPartOffset=2000 , 0
StorageInit ok = 20057
Raw SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit ret = 0, SecureMode = 0
atags_set_bootdev: ret:(0)
GPT 0x337a9f0 signature is wrong
recovery gpt...
GPT 0x337a9f0 signature is wrong
recovery gpt fail!
LoadTrust Addr:0x4000
No find bl30.bin
No find bl32.bin
Load uboot, ReadLba = 2000
hdr 000000000337a3b0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x0                                                                                                                                                             0,0x00,0x00,0x00,0x00,0x00,

Load OK, addr=0x200000, size=0xab6cc
RunBL31 0x40000
INFO:    Preloader serial: 2
NOTICE:  BL31: v1.3(debug):403e0b816
NOTICE:  BL31: Built : 14:13:08, Aug 11 2020
NOTICE:  BL31:Rockchip release version: v1.3
INFO:    ARM GICv2 driver initialized
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 1
INFO:    plat_rockchip_pmu_init: pd status 0xe
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE init                                                                                                                                                             ialization. SMC`s destined for OPTEE will return SMC_UNK
ERROR:   Error initializing runtime service opteed_fast
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

U-Boot 2020.10-orangepi (Feb 22 2022 - 09:53:24 +0800)

Model: Xunlong Orange Pi R1 Plus LTS
DRAM:  1022 MiB
PMIC:  RK8050 (on=0x10, off=0x00)
MMC:   mmc@ff500000: 1
Loading Environment from MMC... MMC Device 0 not found
*** Warning - No MMC card found, using default environment

In:    serial@ff130000
Out:   serial@ff130000
Err:   serial@ff130000
Model: Xunlong Orange Pi R1 Plus LTS
Net:   eth0: ethernet@ff540000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:1...
Found U-Boot script /boot/boot.scr
3341 bytes read in 6 ms (543 KiB/s)
## Executing script at 00500000
Boot script loaded from mmc 1
230 bytes read in 5 ms (44.9 KiB/s)
13064693 bytes read in 571 ms (21.8 MiB/s)
28647936 bytes read in 2562 ms (10.7 MiB/s)
65295 bytes read in 13 ms (4.8 MiB/s)
2698 bytes read in 12 ms (218.8 KiB/s)
Applying kernel provided DT fixup script (rockchip-fixup.scr)
## Executing script at 09000000
Moving Image from 0x2080000 to 0x2200000, end=3df0000
## Loading init Ramdisk from Legacy Image at 06000000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    13064629 Bytes = 12.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 01f00000
   Booting using the fdt blob at 0x1f00000
   Loading Ramdisk to 3d2b8000, end 3df2d9b5 ... OK
   Loading Device Tree to 000000003d23f000, end 000000003d2b7fff ... OK

Starting kernel ...

Loading, please wait...
Starting version 241
Begin: Loading essential drivers ... done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/local-top ... done.
Begin: Running /scripts/local-premount ... Scanning for Btrfs filesystems
done.
Begin: Will now check root file system ... fsck from util-linux 2.33.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk0p1] fsck.ext4 -a -C0 /dev/mmcblk0p1
/dev/mmcblk0p1: clean, 47734/448224 files, 475899/1880692 blocks
done.
done.
Begin: Running /scripts/local-bottom ... done.
Begin: Running /scripts/init-bottom ... done.
[    8.065518] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000007
[    8.066223] CPU: 0 PID: 1 Comm: init Not tainted 5.10.44-rockchip64 #2.1.8
[    8.066838] Hardware name: Xunlong Orange Pi R1 Plus Lts (DT)
[    8.067350] Call trace:
[    8.067591]  dump_backtrace+0x0/0x1f0
[    8.067933]  show_stack+0x18/0x28
[    8.068244]  dump_stack+0xc8/0x120
[    8.068561]  panic+0x16c/0x36c
[    8.068850]  do_exit+0xa8c/0xa90
[    8.069151]  do_group_exit+0x44/0xa0
[    8.069483]  get_signal+0x10c/0x838
[    8.069807]  do_notify_resume+0x274/0x708
[    8.070174]  work_pending+0xc/0x3bc
[    8.070503] SMP: stopping secondary CPUs
[    8.070871] Kernel Offset: disabled
[    8.071192] CPU features: 0x0240002,20002004
[    8.071575] Memory Limit: none
[    8.071870] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000007 ]---
```

Had the same issue a few months earlier and switch the microsd card and that resolved it for a while.

baiywt, 2025-04-08:
Could you please confirm which operating system you're using on the Orange Pi R1 Plus LTS? Is it the official OS or a different distribution?

AlexanderSlaa, 2025-04-08:
official OS