orangepi-xunlong/orangepi-build #98: orangepizero3 build fail

guation, 2023-08-17:
step:
```
clone https://github.com/orangepi-xunlong/orangepi-build.git -b next
cd orangepi-build
sudo ./build.sh  --->  Full OS image for flashing  --->  Show a kernel configuration menu before compilation  --->   orangepizero3 Allwinner H618 quad core 1GB/1.5GB/2GB/4GB RAM WiFi/BT GBE SPI   --->   next    Use the latest kernel   --->   Others 1/2/4 GB Memory   --->   jammy    Ubuntu jammy 22.04 LTS   --->   no  Image with console interface (server)   --->   no  Standard image with console interface
```
system info:
```
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.15.0-79-generic #86-Ubuntu SMP Mon Jul 10 16:07:21 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
ubuntu@ubuntu:~$ cat /etc/os-release 
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```
error log:
```
[ o.k. ] Free SD cache [ 18% ]
[ o.k. ] Mount point 
[ o.k. ] Ending debootstrap process and preparing cache [ jammy ]
[ o.k. ] Unmounting [ /home/ubuntu/orangepi-build/.tmp/rootfs-ba88abc0-042f-4b33-b639-4b38501e6fce ]
jammy-cli-arm64.d4c...0ea.tar.lz4: 1.15GiB [75.0MiB/s] [===================================================================================================================================================================================] 105%
[ o.k. ] Applying distribution specific tweaks for [ jammy ]
[ o.k. ] Applying common tweaks 
[ .... ] Cleaning [ package lists ]
[ .... ] Updating [ package lists ]
[ .... ] Temporarily disabling [ initramfs-tools hook for kernel ]
[ .... ] Installing [ linux-u-boot-next-orangepizero3_1.0.0_arm64.deb ]
[ .... ] Installing [ linux-image-next-sun50iw9_1.0.0_arm64.deb ]
[ .... ] Installing [ linux-dtb-next-sun50iw9_1.0.0_arm64.deb ]
[ .... ] Installing [ orangepi-bsp-cli-orangepizero3_1.0.0_arm64.deb ]
[ .... ] Installing [ orangepi-firmware_1.0.0_all.deb ]
[ .... ] Installing [ orangepi-config_1.0.0_all.deb ]
[ .... ] Installing [ orangepi-zsh_1.0.0_all.deb ]
[ error ] ERROR in function install_deb_chroot [ build.sh:305 -> scripts/main.sh:628 -> scripts/main.sh:597 -> scripts/debootstrap.sh:61 -> scripts/distributions.sh:388 -> scripts/image-helpers.sh:221 -> scripts/general.sh:0 ]
[ error ] Installation of /root/orangepi-zsh_1.0.0_all.deb failed [ orangepizero3 jammy no sun50iw9 ]
[ o.k. ] Process terminated 
[ error ] unmount_on_exit() called! [ build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:0 ]
[ o.k. ] Unmounting [ /home/ubuntu/orangepi-build/.tmp/rootfs-ba88abc0-042f-4b33-b639-4b38501e6fce/ ]
[ error ] ERROR in function unmount_on_exit [ build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:90 -> scripts/general.sh:0 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

JensvandeWiel, 2023-08-18:
Any updates on this? Is support for h618 planned?

guation, 2023-08-18:
> Any updates on this? Is support for h618 planned?

The documentation for OrangePi Zero3 mentions that the source code for building the OrangePi Zero3 image can be obtained from here, and I believe it should support H618. But the reality is that no matter what option is chosen, building will fail.

guation, 2023-08-19:
> Any updates on this? Is support for h618 planned?

However, from the logs, it can be seen that the build failed while synthesizing the system image (rootfs), and the uboot and kernel builds have already been successful. And based on the existing content analysis in DTS, I believe that there seems to be no essential difference between H618 and H616.

guation, 2023-08-24:
The default DNS (1.0.0.1) is not available in mainland China. When starting the compilation script, you need to use `./build.sh NAMESERVER=223.5.5.5` to manually specify the DNS parameters.

closed 2023-08-24 by guation