orangepi-xunlong/orangepi-build #103: OrangePi3B eMMC bootloader issues

beanflyingau, 2023-09-28:
Tried this on several OPi images downloaded from your Google Drive using both DD and RKDevTool.

When the image successfully installs to the drive and you go to apt update/upgrade any subsequent reboot fails and puts you back into a broken bootloader mode. 

This is repeatable and yes I have been clearing the SPIflash before a fresh install attempt.

Currently this makes the 3B unusable from an eMMC.

w568w, 2023-11-19:
Same device, same problem. I can confirm that this happens on **Orange Pi OS (Arch-based)** too.

Starting the OS, running `sudo pacman -Syu`, applying all updates and rebooting, then you are definitely thrown into the uboot emergency shell (through serial)... Until you flash the eMMC with RKDevTool again.

Have you found any solution to this issue?

beanflyingau, 2023-11-19:
Mine is sitting in the corner of the bench in disgrace kind of like the lack of support offered here!

Apart from that the attached potential solution which I need to find time to try is attached from the OPi discord group. In particular if you have 5+ or 5's more generally worth joining as some of the external OS devs are there.

[image]

baiywt, 2025-02-18:
Could you share the serial console log?