orangepi-xunlong/orangepi-build #105: Please update script to Debian 12 (Server) / Ubuntu 23.04 LTS (Server)

Paderman, 2023-10-03:
Please update to Debian 12 (Server) / Ubuntu 23.04 LTS (Server) because older versions are a little bit obsolete.

The script give me only Debian 10 "Buster" and the old Ubuntu from 2018 and 2020.

Thank you :-)

orangepi-xunlong, 2023-11-22:
http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_Zero_3#Linux_SDK.E2.80.94.E2.80.94orangepi-build_instruction

closed 2023-11-22 by orangepi-xunlong

puckettgw, 2024-09-15:
Using the next branch still doesn't allow to build a bookworm image on orangepi 5 plus.

Paderman, 2024-10-08:
Is it possible to build for latest Debian?

I use the officially version of latest Debian, but it seems be a bug, if I want to use Docker... The System shut down after while, all LED's are turned off and I must re-plugged power...