orangepi-xunlong/orangepi-build #107: u-boot build fails

Tarwyn76, 2023-10-04:
Unable to build the u-boot.  
keep getting this error:
ERROR in function compli_atf [build.sh:305 -> scripts/main.sh:629 -> scripts/main.sh513 -> scripts/compilation.sh:51 -> scripts/general.sh:0 ]
Could not find required toolchain [aarch64-none-linux-gnu-gcc >8.0]

orangepi-xunlong, 2023-11-22:
Delete the toolchains folder and try again.

closed 2023-11-22 by orangepi-xunlong

Boyan81, 2024-12-21:
I have the same one as Tarwyn76
[ ok ] Cleaning /home/boyan/orangepi-build/output/debs for [ orangepizero3 next ]
[ ok ] Compiling ATF
[ ok ] Compiler version [ aarch64-none-linux-gnu-gcc 9.2.1 ]
[ ok ] Started patching process for [ atf sun50iw9-orangepizero3-next ]
[ ok ] Looking for user patches in [ userpatches/atf/atf-sun50iw9 ]
 CC bl31/bl31_context_mgmt.c
 CC bl31/bl31_main.c
 CC bl31/bl31_traps.c
 CC bl31/interrupt_mgmt.c
[ error ] ERROR in function compile_atf [ ./build.sh:305 -> scripts/main.sh:638 -> scripts/main.sh:522 -> scripts/compilation.sh:86 -> scripts/general.sh: 0 ]
[ error ] ATF compilation failed
[ ok ] Process terminated

I delete the toolchains folder, create a new one and still the same result.