orangepi-xunlong/orangepi-build #118: orangepi 4 build fail, ERROR in function prepare_host

HeBea, 2023-12-23:
I use ubuntu18.04 virt machine to build image, but it show an error:

[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ bionic ]
[ error ] ERROR in function prepare_host [ ./build.sh:305 -> scripts/main.sh:630 -> scripts/main.sh:435 -> scripts/general.sh:1467 -> scripts/general.sh:0 ]
[ error ] It seems you ignore documentation and run an unsupported build system: bionic
[ o.k. ] Process terminated

HeBea, 2023-12-23:
why the user guide suggest that we should build it on Ubuntu 18.04 but currently build system only support ubuntu 21.04.x host?

user guide shows:
Linux SDK 的编译都是在安装有 Ubuntu 18.04 的 PC 或者虚拟机（VirtualBox
或VMware）上进行的，

the general.sh script shows:
Ubuntu 21.04.x (Hirsute) x86_64 is the only fully supported host OS release

killa1218, 2024-11-29:
Where are the user guides? I can't find any of them... Could you please help me? @HeBea

baiywt, 2025-02-17:
You should use the main branch.
cd orangepi-build
git checkout main