orangepi-xunlong/orangepi-build #119: 官方build构建orangepi zero 3 wifi加载错误

lazybat, 2023-12-25:
dmesg：
` 6.102421] WCN: combin_img 0 marlin_firmware_write finish and successful
[    6.102619] WCN: marlin_start_run read reset reg val:0x1
[    6.102628] WCN: after do marlin_start_run reset reg val:0x0
[    6.102816] WCN: s_marlin_bootup_time=6102815850
[    6.108496] WCN: clock mode: TSX
[    6.108701] WCN: marlin_write_cali_data sync init_state:0x2a1303e0
[    6.130503] WCN: marlin_write_cali_data sync init_state:0x2a1303e0
[    6.152536] WCN: marlin_write_cali_data sync init_state:0xf0f0f0f1
[    6.152571] WCN: sdio_config bt_wake_host trigger:[high]
[    6.152575] WCN: sdio_config irq:[inband]
[    6.152579] WCN: sdio_config wake_host_level_duration_time:[20ms]
[    6.152586] WCN: sdio_config wake_host_data_separation:[bt/wifi reuse]
[    6.152591] WCN: marlin_send_sdio_config_to_cp sdio_config:0x80f01 (enable config)
[    6.152962] WCN: marlin_write_cali_data finish
[    6.153154] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[    6.174536] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[    6.196537] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[    6.218527] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[    6.240531] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[    6.262560] WCN: check_cp_ready sync val:0xf0f0f0f2, prj_type val:0x0
[    6.284531] WCN: check_cp_ready sync val:0xf0f0f0ff, prj_type val:0x0
[    6.284568] sdiohal:sdiohal_runtime_get entry
[    6.284875] WCN: get_cp2_version entry!
[    6.323965] WCN: WCND at cmd read:WCN_VER:Platform Version:MARLIN3_19B_W21.05.3~Project Version:sc2355_marlin3_lite_ott~12-15-2021 11:26:33~
[    6.324017] WCN: switch_cp2_log - close entry!
[    6.325216] WCN: WCND at cmd read:OK
[    6.325260] WCN: then marlin download finished and run ok
[    6.325272] WCN: start_loopcheck
[    6.325394] sprdwl:sprdwl_cmd_init wakeup source register error.
[    6.339851] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
[    6.339904] Mem abort info:
[    6.339908]   ESR = 0x0000000096000044
[    6.339914]   EC = 0x25: DABT (current EL), IL = 32 bits
[    6.339923]   SET = 0, FnV = 0
[    6.339928]   EA = 0, S1PTW = 0
[    6.339933]   FSC = 0x04: level 0 translation fault
[    6.339939] Data abort info:
[    6.339943]   ISV = 0, ISS = 0x00000044
[    6.339948]   CM = 0, WnR = 1
[    6.339953] user pgtable: 4k pages, 48-bit VAs, pgdp=00000000442ce000
[    6.339961] [0000000000000000] pgd=0000000000000000, p4d=0000000000000000
[    6.339980] Internal error: Oops: 0000000096000044 [#1] PREEMPT SMP
[    6.346283] Modules linked in: lz4hc lz4 polyval_ce polyval_generic sunxi_cir rc_core sunxi_cedrus(C) videobuf2_dma_contig dw_hdmi_cec v4l2_mem2mem videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc zram display_connector sprdwl_ng(+) sunxi_addr cfg80211 rfkill uwe5622_bsp_sdio fuse motorcomm dwmac_sun8i mdio_mux
[    6.374989] CPU: 0 PID: 287 Comm: systemd-modules Tainted: G         C         6.1.31-sun50iw9 #1.0.2
[    6.384232] Hardware name: OrangePi Zero3 (DT)
[    6.388697] pstate: 600000c5 (nZCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    6.395683] pc : wait_for_completion_timeout+0xb4/0x158
[    6.400952] lr : wait_for_completion_timeout+0x38/0x158
[    6.406181] sp : ffff80000a113540
[    6.409497] x29: ffff80000a113540 x28: 0000000000000000 x27: ffff000004590000
[    6.416643] x26: ffff800000ea7a90 x25: 0000000000000002 x24: ffff000004590000
[    6.423785] x23: ffff80000a1135a8 x22: ffff000004590000 x21: ffff800000ea7a88
[    6.430927] x20: ffff800000ea7a80 x19: 0000000000000bb8 x18: 0000000000000002
[    6.438069] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[    6.445211] x14: 0000000000000001 x13: 00000000000333c7 x12: 000000000000016b
[    6.452353] x11: 0000000000000001 x10: 0000000000000a20 x9 : 0000000000000000
[    6.459495] x8 : ffff000004b71400 x7 : 0000000000000000 x6 : 0000000006051b1e
[    6.466637] x5 : 00ffffffffffffff x4 : ffffffffffffffff x3 : 0000000000000000
[    6.473778] x2 : 0000000000000001 x1 : 0000000000000000 x0 : 0000000000000000
[    6.480921] Call trace:
[    6.483371]  wait_for_completion_timeout+0xb4/0x158
[    6.488254]  sprdwl_cmd_send_recv+0x130/0x4b0 [sprdwl_ng]
[    6.493743]  sprdwl_sync_version+0x80/0xe0 [sprdwl_ng]
[    6.498941]  sprdwl_core_init+0x28/0x250 [sprdwl_ng]
[    6.503965]  sprdwl_probe+0x208/0x348 [sprdwl_ng]
[    6.508729]  platform_probe+0x68/0xd8
[    6.512400]  really_probe+0xc4/0x3a8
[    6.515980]  __driver_probe_device+0x7c/0x160
[    6.520341]  driver_probe_device+0x44/0x128
[    6.524528]  __device_attach_driver+0xbc/0x158
[    6.528976]  bus_for_each_drv+0x84/0xd8
[    6.532816]  __device_attach+0xf8/0x1b0
[    6.536657]  device_initial_probe+0x20/0x30
[    6.540844]  bus_probe_device+0xa0/0xa8
[    6.544684]  device_add+0x3c0/0x768
[    6.548180]  platform_device_add+0x104/0x270
[    6.552455]  unisoc_wlan_init+0x4c/0x1000 [sprdwl_ng]
[    6.557567]  do_one_initcall+0x58/0x1d8
[    6.561409]  do_init_module+0x48/0x1d0
[    6.565168]  load_module+0x1a28/0x1ea0
[    6.568924]  __do_sys_finit_module+0xc8/0x140
[    6.573287]  __arm64_sys_finit_module+0x24/0x38
[    6.577824]  invoke_syscall+0x58/0x130
[    6.581581]  el0_svc_common.constprop.0+0x58/0x108
[    6.586376]  do_el0_svc+0x38/0xc0
[    6.589697]  el0_svc+0x34/0x108
[    6.592843]  el0t_64_sync_handler+0xb8/0xc0
[    6.597030]  el0t_64_sync+0x18c/0x190
[    6.600701] Code: 54fffe41 f9400e80 f9000e97 a90683fa (f9000017)
[    6.606796] ---[ end trace 0000000000000000 ]---
`

FlameKm, 2024-03-19:
到现在怎么还没解决

lazybat, 2024-05-08:
> 兄弟你解决了吗，我也遇到了这个问题

电源管理不要改动，就能正常编译。关闭某些节电的就不行了。

sesame-max, 2024-11-13:
官方构建的/lib/firmware目录下缺少wcnmodem.bin和wifi_2355b001_1ant.ini这两个文件，从其他官方镜像复制过去就能加载WiFi了