orangepi-xunlong/orangepi-build #123: Enable SATA on Orange Pi 3B

beelze, 2024-01-05:
My M.2 drive (sata, kingspec nt-256, tested working)  is not detected.

@orangepi-xunlong wrote (#53) that _"dts is set to nvme ssd by default, and if using sata ssd please add overlays=ssd-sata to /boot/orangepiEnv.txt"_ but this is a Orange Pi 5 recipe. How to do the same for Orange Pi 3B?

beelze, 2024-01-16:
The only way I've found to enable SATA on Orange Pi 3B was
`fdtfile=rockchip/rk3566-orangepi-3b-sata.dtb` in `/boot/armbianEnv.txt` (Armbian 6.6.2-edge-rockchip64)

Macaquinyo, 2024-09-21:
Still no solution for the orange pi operating system... Please offer a solution to disable nvme and enable SATA.

Macaquinyo, 2024-09-23:
Please any idea to install the system and boot without SD card from SATA SSD @orangepi-xunlong ?

hywax, 2024-11-27:
Is there any news?

svldd, 2024-11-28:
Managed to get SATA working with Armbian_community_25.2.0-trunk.13_Orangepi3b_bookworm_vendor_6.1.75_minimal.img.xz !
Adjusting the overlays with armbian-config didn't worked. 

Helped adding this line to /boot/armbianEnv.txt:

overlays=rk3566-roc-pc-sata2

After reboot I can see and access SATA device:
root@orangepi3b:~# lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda            8:0    0 111.8G  0 disk
├─sda1         8:1    0 110.8G  0 part
├─sda2         8:2    0     1K  0 part
└─sda5         8:5    0   975M  0 part
mtdblock0     31:0    0    16M  0 disk
mmcblk0      179:0    0  58.2G  0 disk
└─mmcblk0p1  179:1    0  57.6G  0 part /var/log.hdd
                                       /
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
zram0        252:0    0   1.9G  0 disk [SWAP]
zram1        252:1    0    50M  0 disk /var/log
zram2        252:2    0     0B  0 disk

Arcanengineer, 2025-03-21:
Ok. Still would love to be able to do that with the orangepi.org Os images. Like the bullseye debian servers. Because the armbian version is bookworm and I think it was very unstable. Anyone know how?

defencedog, 2025-04-11:
@Macaquinyo this SBC is little underpowered. Use Patriot low voltage P300p 256gb NVME for best compatibility

HikVision didn't work for me

voir64, 2025-10-31:
hi
This works for me too but;
Please any idea to install the system and boot without SD card from SATA SSD