orangepi-xunlong/orangepi-build #125: orangepi 5 plus linux kernel build with error

yjy19999, 2024-01-08:
[error.txt](https://github.com/orangepi-xunlong/orangepi-build/files/13861672/error.txt)
The error is too long to post and it is in this file.
One possible reason is the github is connection is not stable but the build script did not indicate this.

yjy19999, 2024-01-08:
I am using ubuntu 22.04 to build the kernel. I have successfully built the uboot

zql77, 2024-01-31:
我也出现构建内核失败了，报错是这样的：arch/arm64/boot/dts/rockchip/rk3566-orangepi-cm4.dts:1290.14-1299.11: Warning (graph_port): /rkisp-vir1/port: graph node unit address error, expected "0"

defencedog, 2024-07-28:
Anyone solved this issue. I am having same issue in building OrangePi-3b kernel module

```
/home/ukhan/orangepi-build/toolchains/gcc-arm-9.2-2019.12-x86_64-aarch64-none-linux-gnu/bin/../lib/gcc/aarch64-none-linux-gnu/9.2.1/plugin/include/system.h:687:10: fatal error: gmp.h: No such file or directory
make[2]: *** [scripts/gcc-plugins/Makefile:47: scripts/gcc-plugins/structleak_plugin.so] Error 1
make[1]: *** [scripts/Makefile.build:516: scripts/gcc-plugins] Error 2
make: *** [Makefile:1326: scripts] Error 2
[ error ] ERROR in function compile_kernel [ ./build.sh:305 -> scripts/main.sh:624 -> scripts/main.sh:527 -> scripts/compilation.sh:506 -> scripts/general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated
```

Pure-Zzzz, 2024-11-26:
> Anyone solved this issue. I am having same issue in building OrangePi-3b kernel module
>
> ``` [...]
try this
`sudo apt-get install libgmp3-dev libmpfr-dev
sudo apt-get install libmpc-dev`

baiywt, 2025-04-08:
I’ve built it on my side without any issues.
[image]