orangepi-xunlong/orangepi-build #126: Default username and password for debian rootfs?

codernavi18, 2024-01-10:
I built debian bookworm rootfs for server version and it gives following prompt at boot:
`lab login:`

What is the default username and password? I tried both minimal and standard image builds and both of them giving same prompt. I tried root/root, root/lab, lab/lab, orangepi/orangepi, oem/oem, but nothing worked.

Troglodyte2, 2024-02-14:
Debian12 Bookworm installed OrangePi 5+ 02_13_24.  Issue is changing from default password.  No menu choice found to do this task.

baiywt, 2025-04-08:
@codernavi18 Did you use an official Orange Pi image? Our Debian images only have ‌root‌ (no password) and ‌orangepi‌ (password: orangepi) users. The lab user doesn’t exist by default.