orangepi-xunlong/orangepi-build #131: Uboot build error for Orange pi 3b

thiencuong, 2024-02-05:
i meet error when build orange pi 3b. please check.

  OBJCOPY u-boot-nodtb.bin
make[2]: 'arch/arm/dts/rk3566-orangepi-3b.dtb' is up to date.
  COPY    u-boot.dtb
start=$(ccache aarch64-none-linux-gnu-nm u-boot | grep __rel_dyn_start | cut -f 1 -d ' '); end=$(ccache aarch64-none-linux-gnu-nm u-boot | grep __rel_dyn_end | cut -f 1 -d ' '); tools/relocate-rela u-boot-nodtb.bin 0x00a00000 $start $end
  MKIMAGE u-boot.itb
  CC      spl/common/spl/spl.o
  CC      spl/arch/arm/cpu/armv8/fwcall.o
  LD      spl/arch/arm/cpu/armv8/built-in.o
  CC      spl/lib/display_options.o
  LD      spl/common/spl/built-in.o
  LD      spl/lib/built-in.o
  LD      spl/u-boot-spl
  OBJCOPY spl/u-boot-spl-nodtb.bin
  CAT     spl/u-boot-spl-dtb.bin
  COPY    spl/u-boot-spl.bin
[ error ] ERROR in function compile_uboot [ build.sh:305 -> scripts/main.sh:630 -> scripts/main.sh:516 -> scripts/compilation.sh:256 -> scripts/general.sh:0 ]
[ error ] U-boot compilation failed 
[ o.k. ] Process terminated

thiencuong, 2024-03-01:
Hi Orange PI team. Is there any update for this issue?

baiywt, 2025-02-17:
Did you compile it on an Ubuntu 22.04 host?
Please provide your output/debug/compilation.log

closed 2025-03-17 by orangepi-xunlong