orangepi-xunlong/orangepi-build #132: Orange Pi OS (Arch) for Orange Pi 5 Plus comes with outdated pacman mirror (http://nl.mirror.orangeinfra.online/)

archalchemy, 2024-02-05:
I would like to report this problem that happens when I want to update the OS, the repository seems to be unavailable. Is there any other repository that can be used?

# pacman -Syu
:: Synchronizing package databases...
 core.db failed to download
 extra.db failed to download
 community.db failed to download
error: failed retrieving file 'core.db' from nl.mirror.orangeinfra.online : Failed to connect to nl.mirror.orangeinfra.online port 80 after 10000 ms: Timeout was reached
error: failed retrieving file 'extra.db' from nl.mirror.orangeinfra.online : Failed to connect to nl.mirror.orangeinfra.online port 80 after 10000 ms: Timeout was reached
error: failed retrieving file 'community.db' from nl.mirror.orangeinfra.online : Failed to connect to nl.mirror.orangeinfra.online port 80 after 10000 ms: Timeout was reached
warning: too many errors from nl.mirror.orangeinfra.online, skipping for the remainder of this transaction
error: failed to synchronize all databases (download library error)

MarceloJads, 2024-02-08:
I was going through the same problem with my Orange Pi Zero 2. 

I found some mirrors that work (at least it can install some packages). The mirror list is within the following link: [https://archlinuxarm.org/about/mirrors](url).

By running `rankmirrors` command, I was able to order some of them in terms of latency:

```
Server = http://de4.mirror.archlinuxarm.org/$arch/$repo
Server = http://jp.mirror.archlinuxarm.org/$arch/$repo
Server = https://de3.mirror.archlinuxarm.org/$arch/$repo
Server = https://eu.mirror.archlinuxarm.org/$arch/$repo
Server = https://fl.us.mirror.archlinuxarm.org/$arch/$repo
Server = https://mirrors.dotsrc.org/archlinuxarm/$arch/$repo
Server = https://uk.mirror.archlinuxarm.org/$arch/$repo
```
You can copy these into the file `/etc/pacman.d/mirrorlist`.

ms32035, 2024-08-10:
@archalchemy in the latest os image I found something like this, which seems to work
```
[opios-core]
Server = https://mirror.orangepi.dev/archlinux/stable/$arch/$repo

[opios-extra]
Server = https://mirror.orangepi.dev/archlinux/stable/$arch/$repo
```