orangepi-xunlong/orangepi-build #135: 构建时报错： ERROR in function install_deb_chroot

kwanxian, 2024-02-16:
当前分支：next - 55155f1d
构建目标：orangepizero2w (1GB), server standard/minimal
构建环境：ubuntu 22.04.3
```
[ error ] ERROR in function install_deb_chroot [ ./build.sh:305 -> scripts/main.sh:637 -> scripts/main.sh:606 -> scripts/debootstrap.sh:64 -> scripts/distributions.sh:388 -> scripts/image-helpers.sh:221 -> scripts/general.sh:0 ]
[ error ] Installation of /root/orangepi-zsh_1.0.0_all.deb failed [ orangepizero2w jammy no sun50iw9 ]
[ o.k. ] Process terminated
[ error ] unmount_on_exit() called! [ ./build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:0 ]
[ o.k. ] Unmounting [ /root/orangepi-build/.tmp/rootfs-6a0740f5-272c-429b-9843-085c63e21ae9/ ]
[ error ] ERROR in function unmount_on_exit [ ./build.sh:305 -> scripts/main.sh:1 -> scripts/image-helpers.sh:90 -> scripts/general.sh:0 ]
[ error ] debootstrap-ng was interrupted
[ o.k. ] Process terminated
```
deb构建成功的，但一直无法安装
[image]

kwanxian, 2024-02-21:
经过多次测试：
更换源
```
vi userpatches/config-example.conf
CHINA_DOWNLOAD_MIRROR="tsinghua"
```

更改NAMESERVER地址
```
./build.sh NAMESERVER=223.5.5.5
```

output/debug/install.log 错误提示
```
W: Failed to fetch http://repo.huaweicloud.com/ubuntu-ports/dists/jammy/InRelease  Could not resolve 'repo.huaweicloud.com'
W: Failed to fetch http://repo.huaweicloud.com/ubuntu-ports/dists/jammy-security/InRelease  Could not resolve 'repo.huaweicloud.com'
W: Failed to fetch http://repo.huaweicloud.com/ubuntu-ports/dists/jammy-updates/InRelease  Could not resolve 'repo.huaweicloud.com'
W: Failed to fetch http://repo.huaweicloud.com/ubuntu-ports/dists/jammy-backports/InRelease  Could not resolve 'repo.huaweicloud.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

Q1anWan, 2024-03-12:
我现在使用国内镜像(userpatches/config-default.conf 中 CHINA_DOWNLOAD_MIRROR具有值时)，为CM4编译时在最终的Installing [xxx.deb] 阶段也出现了大量unable to fetch error

我通过将CHINA_DOWNLOAD_MIRROR注释掉解决了这个问题。推测可能是国内源适配问题

Q1anWan, 2024-03-12:
One more thing: I've used a proxy.

kwanxian, 2024-03-14:
> 我现在使用国内镜像(userpatches/config-default.conf 中 CHINA_DOWNLOAD_MIRROR具有值时)，为CM4编译时在最终的Installing [xxx.deb] 阶段也出现了大量unable to fetch error
> 
> 我通过将CHINA_DOWNLOAD_MIRROR注释掉解决了这个问题。推测可能是国内源适配问题

改成ustc源就可以了

rdyWillBetter, 2024-03-15:
请问要换成ustc源具体需要怎么操作？直接修改/etc/apt/source.list文件吗？

kwanxian, 2024-03-17:
> 请问要换成ustc源具体需要怎么操作？直接修改/etc/apt/source.list文件吗？

scripts/configuration.sh 新增：
```
if [[ ${CHINA_DOWNLOAD_MIRROR} == ustc ]]; then
		DEBIAN_MIRROR='mirrors.ustc.edu.cn/debian'
		DEBIAN_SECURTY='mirrors.ustc.edu.cn/debian-security'
		UBUNTU_MIRROR='mirrors.ustc.edu.cn/ubuntu-ports/'
	fi
```

userpatches/config-example.conf
```
KERNEL_CONFIGURE=""			# leave empty to select each time, set to "yes" or "no" to skip dialog prompt
CLEAN_LEVEL="debs,oldcache"		# comma-separated list of clean targets: "make" = make clean for selected kernel and u-boot,
					# "debs" = delete packages in "./output/debs" for current branch and family,
					# "alldebs" = delete all packages in "./output/debs", "images" = delete "./output/images",
					# "cache" = delete "./output/cache", "sources" = delete "./sources"
					# "oldcache" = remove old cached rootfs except for the newest 8 files

DEST_LANG="en_US.UTF-8"			# sl_SI.UTF-8, en_US.UTF-8

# advanced
EXTERNAL_NEW="prebuilt"			# compile and install or install prebuilt additional packages
INSTALL_HEADERS=""			# install kernel headers package
LIB_TAG="master"			# change to "branchname" to use any branch currently available.
USE_TORRENT="yes"			# use torrent network for faster toolchain and cache download
DOWNLOAD_MIRROR="china"			# set to "china" to use mirrors.tuna.tsinghua.edu.cn

BOARD=""
BRANCH=""
RELEASE=""
WIREGUARD="no"
BUILD_KSRC="no"					
INSTALL_KSRC="no"				
IGNORE_UPDATES="yes"
COMPRESS_OUTPUTIMAGE="no"
NO_APT_CACHER="no"
CHINA_DOWNLOAD_MIRROR="ustc"
#install_balena_etcher="yes"
#install_zfs="yes"
#install_docker="yes"
#install_chromium="yes"
#install_firefox="yes"

```

baiywt, 2025-04-08:
@kwanxian 你好，我测了下没出现这个问题。
[image]