orangepi-xunlong/orangepi-build #137: orangepi 5Plus uart serial loop test

13037639065, 2024-02-20:
### 平台环境
硬件：orangepi 5Plus
镜像：Orangepi5plus_1.0.8_ubuntu_focal_desktop_xfce_linux5.10.160.img
内核：Linux D152402102788IY 5.10.160-rockchip-rk3588 #1.0.8 SMP Fri Nov 10 17:58:32 CST 2023 aarch64 aarch64 aarch64 GNU/Linux

测试过程发现uart存在误码率
所以编写一段代码循环发送 0x00 ~ 0xff
测试发现误码存在一定的规律性。
将串口进行短接做回环测试。
发现使用 8N1 115200 测试 /dev/ttyS1发现如下现象
1. 收到的数据会多出一个0随后收到63字节正确数据，随后丢一字节数据。 （每次误码都是这个规律）
2. 并且多个串口测试，误码问题是同时发生的
``` bash
python3 loop.py -p /dev/ttyS4 -b 115200 -r 0
```
测试脚本如下
``` python3
import serial
import threading
import time
import sys
import signal
import argparse

# 初始化统计变量
sent_count = 0
received_count = 0
error_count = 0

# 线程锁，用于同步对全局统计变量的访问
g_running = True

def signal_handler(sig, frame):
    global sent_count, received_count, error_count, g_running
    print("\nExiting...")
    print(f"Sent bytes: {sent_count}, Received bytes: {received_count}, Errors: {error_count}")
    if received_count > 0:
        error_rate = (error_count / received_count) * 100
        print(f"Error rate: {error_rate:.2f}%")
    g_running = False
    sys.exit(0)

def sender(port, baudrate, frequency):
    ser = serial.Serial(port, baudrate=baudrate, timeout=0.5)
    global sent_count,g_running
    
    send_bytes = b''
    for e in range(0, 256):
        send_bytes += int(e).to_bytes(1,"little")
    print(send_bytes.hex())
    while g_running:
        ser.write(send_bytes)
        sent_count += len(send_bytes)
        if frequency !=0:
            time.sleep(1.0/frequency)
    print("sender: exit...")

def receiver(port, baudrate):
    ser = serial.Serial(port, baudrate=baudrate, timeout=0.5)
    print("start receiver")
    global received_count, error_count, g_running
    last_elem = None
    index = 0
    while g_running:
        data = ser.read(ser.in_waiting)
        received_count += len(data)
        for e in data:
            if last_elem == None:
                pass
            else:
                if (last_elem + 1) & 0xff != e:
                    time_str = time.strftime("%Y-%m-%d %H:%M:%S")
                    print(f"{time_str}: data error!!!!! [{last_elem} + 1 != {e}] index: {index}")
                    error_count += 1
            last_elem = e
            index += 1
            
    print("receiver: exit...")

def main():
    parser = argparse.ArgumentParser(description="Serial Port Self-loopback Test")
    parser.add_argument("--port", "-p", help="Serial port", default="/dev/ttyUSB0")
    parser.add_argument("--baudrate", "-b", type=int, help="Baud rate", default=115200)
    parser.add_argument("--frequency", "-r", type=float, help="Sending frequency in Hz", default=200.0)

    args = parser.parse_args()

    # 设置信号处理器
    signal.signal(signal.SIGINT, signal_handler)
    
    print(f"Serial info: {args.port} baudrate: {args.baudrate}")
    print(time.strftime("%Y-%m-%d %H:%M:%S"))

    # 创建并启动发送和接收线程
    sender_thread = threading.Thread(target=sender, args=(args.port, args.baudrate, args.frequency))
    receiver_thread = threading.Thread(target=receiver, args=(args.port, args.baudrate))

    receiver_thread.start()
    sender_thread.start()

    sender_thread.join()
    receiver_thread.join()

if __name__ == "__main__":
    main()

```

13037639065, 2024-02-28:
用程序自环发送数据 `0xa5 0xa5`
间隔一段时间会发生误码，使用示波器抓取到的波形如图
串口配置8N1 115200 
[image]

baiywt, 2025-04-08:
用实时内核看下有没有这个问题，添加下面的编译参数，然后重新编译替换内核。
./build.sh build_rt_image=yes

然后用cyclictest工具测下实时性能，这个没问题的话，再测下串口看看。
[image]