orangepi-xunlong/orangepi-build #139: Ethernet stops working on many Orange Pi 3b units.

guerillaengineer, 2024-02-27:
Please see: https://old.reddit.com/r/OrangePI/comments/1aojmn3/orange_pi_3b_ethernet_doesnt_work/ for the conversation so far.

Something is happening to dozens of people's 3b units, it seems to be related to using the official images.

If it is the firmware being updated to a faulty release, it would make sense that other OSes tried after would continue having this issue, which is the case for many.

mizegit, 2024-03-08:
https://www.reddit.com/r/OrangePI/comments/1aojmn3/comment/ktqtmle/

> run as root
> 
> io -4 0xFDC60284 0x3f3f3f3f
> 
> io -4 0xFDC6028C 0x003f003f
> 
> io -4 0xFDC60298 0x3f003f00
> 
> io -4 0xFDC6029C 0x3f3f3f3f
> 
> io -4 0xfdc60388 0xFFFF0049

guerillaengineer, 2024-03-08:
Oops! yeah sorry, I'm Umbra in the discord server, I forgot to close this after Yang found the fix. Thanks for reminding me!

> https://www.reddit.com/r/OrangePI/comments/1aojmn3/comment/ktqtmle/
> 
> > run as root
> > io -4 0xFDC60284 0x3f3f3f3f
> > io -4 0xFDC6028C 0x003f003f
> > io -4 0xFDC60298 0x3f003f00
> > io -4 0xFDC6029C 0x3f3f3f3f
> > io -4 0xfdc60388 0xFFFF0049

closed 2024-03-08 by guerillaengineer

SuperKali, 2024-03-15:
i have the same problem  :(

guerillaengineer, 2024-03-16:
> i have the same problem :(

it's in a functional state running the command as above. run as root, don't restart or you need to do it again.
has worked for everyone else that has this issue.

nemanjam, 2024-10-24:
How to run `io` command in Ubuntu 24.04?

```bash
username@pi:~/Desktop$ sudo io -4 0xFDC60284 0x3f3f3f3f
sudo: io: command not found

```

MichaIng, 2026-01-01:
If anyone lands here, the `io` command is a blob from Xunlong: https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/packages/bsp/rk356x/usr/local/bin/io

I could not find sources and there is no info what it it does exactly, hence do on your own risk, but can be downloaded to other distros like that:
```sh
sudo wget 'https://raw.githubusercontent.com/orangepi-xunlong/orangepi-build/refs/heads/next/external/packages/bsp/rk356x/usr/local/bin/io' -O /usr/local/bin/io
sudo chmod +x /usr/local/bin/io
```