orangepi-xunlong/orangepi-build #142: zero 2w enable CONFIG_KEYBOARD_MATRIX (and gpio-matrix-keypad)

Coderah, 2024-03-07:
No reason not to have the kernel module for matrix-keypad (https://github.com/torvalds/linux/blob/master/drivers/input/keyboard/matrix_keypad.c) be built and I believe CONFIG_KEYBOARD_MATRIX just needs to be enabled to achieve that?

Coderah, 2024-03-07:
specifically hoping to achieve this in external/config/kernel/linux-6.1-next.config

Coderah, 2024-03-07:
further info https://cateee.net/lkddb/web-lkddb/KEYBOARD_MATRIX.html this is commonly used with other SBCs here is an example https://blog.geggus.net/2017/08/a-matrix-keypad-on-a-raspberry-pi-done-right/

Coderah, 2024-03-07:
if someone can help point me to what generates those config files I'll happily open a PR

baiywt, 2025-04-08:
@Coderah You can modify the external/config/kernel/linux-6.1-sun50iw9-next.config file to enable the CONFIG_KEYBOARD_MATRIX option.