orangepi-xunlong/orangepi-build #143: I2C bus unit address format error, expected "10"

vasilevrv, 2024-03-11:
Hello.

I get an error when building:

```
"arch/arm64/boot/dts/rockchip/rk3588s-orangepi-5-pro.dts" [New DIRECTORY]                                                                                                                                                                                                       0,0-1         All
  CC [M]  fs/xfs/xfs_qm.o
  CC [M]  fs/xfs/xfs_quotaops.o
  CC [M]  fs/xfs/xfs_rtalloc.o
  CC [M]  fs/xfs/xfs_acl.o
  CC [M]  fs/xfs/xfs_sysctl.o
  CC [M]  fs/xfs/xfs_ioctl32.o
  CC [M]  fs/xfs/xfs_pnfs.o
  LD [M]  fs/xfs/xfs.o
arch/arm64/boot/dts/rockchip/rk3588s-orangepi-5-pro.dts:711.20-722.4: Warning (i2c_bus_reg): /i2c@feab0000/es8388@11: I2C bus unit address format error, expected "10"
kernel/sched/core.c:10448:17: internal compiler error: Illegal instruction
0x1732f96 internal_error(char const*, ...)
make[3]: *** [scripts/Makefile.build:250: kernel/sched/core.o] Error 1
make[2]: *** [scripts/Makefile.build:500: kernel/sched] Error 2
make[1]: *** [scripts/Makefile.build:500: kernel] Error 2
make: *** [Makefile:2012: .] Error 2
[ error ] ERROR in function compile_kernel [ ./build.sh:305 -> scripts/main.sh:637 -> scripts/main.sh:540 -> scripts/compilation.sh:515 -> scripts/general.sh:0 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated
```

Target device: Orange PI 5 Plus
Target OS: Ubuntu 22.04 (with Gnome)
Host OS:  Ubuntu 22.04

jcdutton, 2024-05-06:
For this bit:
arch/arm64/boot/dts/rockchip/rk3588s-orangepi-5-pro.dts:711.20-722.4: Warning (i2c_bus_reg): /i2c@feab0000/es8388@11: I2C bus unit address format error, expected "10"

Your item name is "es8388@11"   (the 11 is in hex) so it was expecting the reg field to also contain 0x11, but it has 0x10 in the reg field.
The two numbers need to match.

That does not affect your "compiler error" message. There is a bug in the compiler itself. Try to use a different compiler. E.g. a more modern gcc or clang.
It might help if you mention which CPU you are compiling this on and if you are cross compiling.

baiywt, 2025-04-08:
@vasilevrv Could you confirm if you're building the kernel on a PC running Ubuntu 22.04? Also, could you let me know if your system has enough memory and how much RAM is available during the build process?