orangepi-xunlong/orangepi-build #145: opi5 plus pulseaudio failed to start when lock autologin.

13037639065, 2024-03-15:
硬件：orangepi 5Plus
镜像：Orangepi5plus_1.0.8_ubuntu_focal_desktop_xfce_linux5.10.160.img
内核：Linux D152402102788IY 5.10.160-rockchip-rk3588 https://github.com/orangepi-xunlong/orangepi-build/pull/1.0.8 SMP Fri Nov 10 17:58:32 CST 2023 aarch64 aarch64 aarch64 GNU/Linux

关闭autologin
rm /etc/lightdm/lightdm.conf.d/22-orangepi-autologin.conf
再次重启输入密码。
声卡功能异常

[image]

kingsimba, 2024-03-27:
之前复现步骤没有写清楚。是需要创建一个新的用户，并用这个用户登录，才能复现问题。

原因是 orangepi 和 新创建的用户同时登录，会有多个 pulseaudio 进程抢占声音设备。

解决方法：禁用 orangepi 用户，让它无法登录。

13037639065, 2024-08-25:
pluseaudio config error

closed 2024-08-25 by 13037639065