orangepi-xunlong/orangepi-build #146: OrangePI 3B Ethernet Stopped working

SuperKali, 2024-03-15:
Hello,

I'm reaching out to inform you that I've tested all versions, and unfortunately, after 20 days of utilizing this board, the Ethernet port has ceased to function. It is unable to assign or retrieve DHCP, and configuring the IP address statically has also proven unsuccessful.

Could you please advise on how to rectify this issue?

Best regards,
Dany

ProKn1fe, 2024-03-15:
https://github.com/orangepi-xunlong/orangepi-build/issues/139#issuecomment-1984961934

SuperKali, 2024-03-15:
Same problem

mizegit, 2024-03-16:
It seems like a hardware problem, after long time running, the Ethernet may broken.

SuperKali, 2024-03-16:
I bought it in AliExpress but the seller want that I send the item back in china, here in Italy the shipping cost is very expensive and I don’t know how to do

ProKn1fe, 2024-03-16:
> It seems like a hardware problem, after long time running, the Ethernet may broken.

Still it can be software fixed.

SuperKali, 2024-03-16:
How @ProKn1fe 🥺

ProKn1fe, 2024-03-16:
DId you even open link i posted? It has everything.

SuperKali, 2024-03-16:
I have run those command but still not working

guerillaengineer, 2024-03-16:
just run this as root and DO NOT restart or you have to re-do

io -4 0xFDC60284 0x3f3f3f3f
io -4 0xFDC6028C 0x003f003f
io -4 0xFDC60298 0x3f003f00
io -4 0xFDC6029C 0x3f3f3f3f
io -4 0xfdc60388 0xFFFF0049

SuperKali, 2024-03-16:
How I already told, this not fix the problem, I tried many time but the problem persists.

guerillaengineer, 2024-03-16:
> How I already told, this not fix the problem, I tried many time but the problem persists.

you must have a different issue then. go on the orangepi discord, find Umbra in the orangepi3b channel, dm him  and it will go from there. while you wait on him, type out exactly which distro you're on and what you've been typing into the terminal. screenshots help but aren't necessary

SuperKali, 2024-03-17:
Can you link the discord server?

masterct88, 2024-03-17:
Hi @guerillaengineer, sorry to bother you but maybe i miss something: why i need to run that code you posted before every time i need to restart the OrangePi 3b?
There is some type of fix in development?

yoni13, 2024-03-18:
Hello everyone,legengyang here.
They did promise to fix it on their side, but they haven't built a new build that the Internet works without entering the command.
I will ask them if there will be a new build.

yoni13, 2024-03-19:
Legengyang here.
They replied "Not sure yet"
_目前还不确定。_

SuperKali, 2024-03-21:
> just run this as root and DO NOT restart or you have to re-do
> 
> io -4 0xFDC60284 0x3f3f3f3f io -4 0xFDC6028C 0x003f003f io -4 0xFDC60298 0x3f003f00 io -4 0xFDC6029C 0x3f3f3f3f io -4 0xfdc60388 0xFFFF0049

[image]
[image]

Still not work :(

denis1482, 2024-03-26:
You may have booted an image with wrongly composed DTB and now your ethernet port is damaged:
[image]

SuperKali, 2024-03-26:
@denis1482 I always used the stock image from the orangepi repository.... now what I can do? I contacted the seller and he don't want replace my board...

mizegit, 2024-03-27:
> @denis1482 I always used the stock image from the orangepi repository.... now what I can do? I contacted the seller and he don't want replace my board...

It's not the software problem, someone use the stock image, the Ethernet is also broken.
I know a orangepi 3b user replace the rk3566 and network is fixed, but after several days running,
the Ethernet is broken again.

masterct88, 2024-06-12:
Hi everyone, any news? 
I solved this issue writing this code:
```
io -4 0xFDC60284 0x3f3f3f3f
io -4 0xFDC6028C 0x003f003f
io -4 0xFDC60298 0x3f003f00
io -4 0xFDC6029C 0x3f3f3f3f
io -4 0xfdc60388 0xFFFF0049
exit 0
```
in my /etc/rc.local
Anyone know if it will be fixed or merged?

saki92, 2024-08-24:
> Hi everyone, any news? I solved this issue writing this code:
> 
> ```
> io -4 0xFDC60284 0x3f3f3f3f
> io -4 0xFDC6028C 0x003f003f
> io -4 0xFDC60298 0x3f003f00
> io -4 0xFDC6029C 0x3f3f3f3f
> io -4 0xfdc60388 0xFFFF0049
> exit 0
> ```
> 
> in my /etc/rc.local Anyone know if it will be fixed or merged?

This solved the ethernet problem on my 3b and its working again. Thanks!

nemanjam, 2024-10-24:
> This solved the ethernet problem on my 3b and its working again. Thanks!

How do I run `io` command in Ubuntu when its not defined?

nemanjam, 2024-10-24:
> > > > Hi everyone, any news? I solved this issue writing this code:
> > > > ```
> > > > io -4 0xFDC60284 0x3f3f3f3f
> > > > io -4 0xFDC6028C 0x003f003f
> > > > io -4 0xFDC60298 0x3f003f00
> > > > io -4 0xFDC6029C 0x3f3f3f3f
> > > > io -4 0xfdc60388 0xFFFF0049
> > > > exit 0
> > > > ```
> > > > 
> > > > 
> > > >     
> > > >       
> > > >     
> > > > 
> > > >       
> > > >     
> > > > 
> > > >     
> > > >   
> > > > in my /etc/rc.local Anyone know if it will be fixed or merged?
> > > 
> > > 
> > > This solved the ethernet problem on my 3b and its working again. Thanks!
> > 
> > 
> > How do I run `io` command in Ubuntu when its not defined?
> 
> You probably are not running the official ubuntu from OrangePi. Are you?

This one:

https://joshua-riek.github.io/ubuntu-rockchip-download/boards/orangepi-3b.html

saki92, 2024-10-24:
> > > > > Hi everyone, any news? I solved this issue writing this code:
> > > > > ```
> > > > > io -4 0xFDC60284 0x3f3f3f3f
> > > > > io -4 0xFDC6028C 0x003f003f
> > > > > io -4 0xFDC60298 0x3f003f00
> > > > > io -4 0xFDC6029C 0x3f3f3f3f
> > > > > io -4 0xfdc60388 0xFFFF0049
> > > > > exit 0
> > > > > ```
> > > > > 
> > > > > 
> > > > >     
> > > > >       
> > > > >     
> > > > > 
> > > > >       
> > > > >     
> > > > > 
> > > > >     
> > > > >   
> > > > > in my /etc/rc.local Anyone know if it will be fixed or merged?
> > > > 
> > > > 
> > > > This solved the ethernet problem on my 3b and its working again. Thanks!
> > > 
> > > 
> > > How do I run `io` command in Ubuntu when its not defined?
> > 
> > 
> > You probably are not running the official ubuntu from OrangePi. Are you?
> 
> This one:
> 
> https://joshua-riek.github.io/ubuntu-rockchip-download/boards/orangepi-3b.html
> 
> Official Orange Ubuntu Ethernet works fine, but graphic is laggy.
> 
> ![image](https://private-user-images.githubusercontent.com/9990165/379901643-032384be-24dc-40b8-ac9c-b01a42e676b9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk4MDg5NDcsIm5iZiI6MTcyOTgwODY0NywicGF0aCI6Ii85OTkwMTY1LzM3OTkwMTY0My0wMzIzODRiZS0yNGRjLTQwYjgtYWM5Yy1iMDFhNDJlNjc2YjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjRUMjIyNDA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDQ0Nzg2YmFiNmE2MzIxZGUxMjFiZDhkMjgzYTE5MGVjOTZkYTg0NjNmNTQyNjgwNDQ2MDdlNWYyNjQ1ZGU0MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.0oGEEOAoWkmqXK7lU09VCG4xK7BQbtN3VcuU7JkQdDQ)

I don't know about these images. I am using the one from the google drive and the commands worked. The ethernet actually worked fine for months. Then one day it stopped working and got it working again after writing these values to the addresses.

nemanjam, 2024-10-25:
Ethernet works fine on the official image but UI is very laggy, video rendering is by far slower than other distros and also kernel is outdated.

yoni13, 2024-10-25:
I am not an expert in this but have you tried to install the official image and install third party kernel?
IDK if that works though

alglez, 2025-01-30:
I'm experiencing the same problem with arch based orangepi os. There is no io tool in arch, and never used any other image than the official ones.

In my case ethernet worked but unreliable if i limit the speed, affecting even wlan ip if i connect through ssh to the wifi ip (not ethernet) but both connections are established, the only solution is to connect just the wlan and remove any cable from the ethernet port.