orangepi-xunlong/orangepi-build #149: OrangePi 构建时下载完成资源后报错异常停止

FlameKm, 2024-03-19:
最近（2024.3.19）才有的问题，构建选项
Full OS image for flashing    
Do not change the kernel configuration
orangepizero3
next
Others
Ubuntu
no
yes

后面尝试只选择内核一样报错
```
⇒  ./build.sh
[ warn ] This script requires root privileges, trying to use sudo 
[ o.k. ] Using config file [ /home/hyc/Project/orangepi-build/userpatches/config-example.conf ]
[ .... ] Extension being added [ sunxi-tools :: added by build.sh:305 -> scripts/main.sh:384 -> scripts/configuration.sh:151 -> external/config/sources/families/sun50iw9.conf:1 -> external/config/sources/families/include/sunxi64_common.inc:1 -> scripts/extensions.sh:0 ]
[ o.k. ] Extension manager [ processed 3 Extension Methods calls and 3 Extension Method implementations ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ jammy ]
[ .... ] Installing build dependencies 
[ o.k. ] Syncing clock [ cn.pool.ntp.org ]
[ o.k. ] Checking for external GCC compilers 
[ o.k. ] Downloading sources 
[ o.k. ] Checking git sources [ /home/hyc/Project/orangepi-build/u-boot v2021.07-sunxi ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ /home/hyc/Project/orangepi-build/kernel orange-pi-6.1-sun50iw9 ]
[ .... ]  Cleaning ....  [ 1 files ]
[ o.k. ] Checking git sources [ /home/hyc/Project/orangepi-build/external/cache/sources/arm-trusted-firmware-sunxi-mainline master ]
[ .... ] Up to date 
[ o.k. ] Checking git sources [ /home/hyc/Project/orangepi-build/external/cache/sources/sunxi-tools master ]
[ .... ] Up to date 
[ o.k. ] Cleaning /home/hyc/Project/orangepi-build/output/debs for [ orangepizero3 next ]
[ o.k. ] Compiling ATF 
[ o.k. ] Compiler version [ aarch64-none-linux-gnu-gcc 9.2.1 ]
[ o.k. ] Started patching process for [ atf sun50iw9-orangepizero3-next ]
[ o.k. ] Looking for user patches in [ userpatches/atf/atf-sun50iw9 ]
  AR      /home/hyc/Project/orangepi-build/external/cache/sources/arm-trusted-firmware-sunxi-mainline/master/build/sun50i_h616/debug/lib/libfdt.a
  AR      /home/hyc/Project/orangepi-build/external/cache/sources/arm-trusted-firmware-sunxi-mainline/master/build/sun50i_h616/debug/lib/libc.a
[ error ] ERROR in function compile_atf [ build.sh:305 -> scripts/main.sh:637 -> scripts/main.sh:521 -> scripts/compilation.sh:86 -> scripts/general.sh:0 ]
[ error ] ATF compilation failed 
[ o.k. ] Process terminate
```

pessimism, 2024-03-21:
Same error at same place after choosing:
Full OS image for flashing
Do not change the kernel configuration
orangepizero2
bookworm
yes
xfce
base
select 3dsupport

build host is fresh install ubuntu 22.04

[ o.k. ] Cleaning /home/steve/orangepi-build/output/debs for [ orangepizero2 next ]
[ o.k. ] Compiling ATF
[ o.k. ] Compiler version [ aarch64-none-linux-gnu-gcc 9.2.1 ]
[ o.k. ] Started patching process for [ atf sun50iw9-orangepizero2-next ]
[ o.k. ] Looking for user patches in [ userpatches/atf/atf-sun50iw9 ]
  AR      /home/steve/orangepi-build/external/cache/sources/arm-trusted-firmware-sunxi-mainline/master/build/sun50i_h616/debug/lib/libfdt.a
  AR      /home/steve/orangepi-build/external/cache/sources/arm-trusted-firmware-sunxi-mainline/master/build/sun50i_h616/debug/lib/libc.a
[ error ] ERROR in function compile_atf [ build.sh:305 -> scripts/main.sh:637 -> scripts/main.sh:521 -> scripts/compilation.sh:86 -> scripts/general.sh:0 ]
[ error ] ATF compilation failed
[ o.k. ] Process terminated

coinight, 2024-03-29:
same,can't build

nrsandoval, 2024-03-30:
Armbian ran into the same [issue](https://github.com/armbian/build/pull/3790). You'll want to change the `external/config/sources/families/*.conf` for your specific chip. For an orangepizero2, it'll be `sun50iw9.conf`, and you'll change the `branch:master` for ATFBRANCH to the commit id from the issue.

zcy618, 2024-03-31:
the same issue, what should I do?

nrsandoval, 2024-04-05:
Hi everyone, I've attached a patch that I've been using to compile and so far it seems to work. Just patch in the orangepi-build directory and it should fix orangepi zero 2 image builds.

2 notes:
1. The commit id is fairly old, and I'm not sure how much that might negatively affect the image. If someone can find a more recent commit id that works it would be greatly appreciated.
2. This patch works for orangepi zero 2 images, if your issue is on a different board, you'll need to find which `external/config/sources/families/*.conf` file your board uses and make the same change as the patch.

[atf_branch.patch](https://github.com/orangepi-xunlong/orangepi-build/files/14888126/atf_branch.patch)

leeboby, 2025-02-17:
已修复，谢谢！

closed 2025-03-17 by orangepi-xunlong

ytech-cn, 2025-03-19:
这个怎么解决的？