orangepi-xunlong/orangepi-build #151: orangepi 3b 构建 rootfs 在安装 zsh 包时会卡住，需要回车才能继续

Alex6357, 2024-03-31:
如题，可能是后台在等待确认，建议显示全部输出，能更好的知道正在进行什么步骤，是等待操作还是构建卡住。

tianfeng-yang, 2024-07-19:
遇到相同问题

tianfeng-yang, 2024-07-31:
我找到这个问题了，可以通过`output/debug/install.log`查看，挂在了安装`libpulse0:arm64`包时，已经存在`/etc/pulse/client.conf`这个问题，需要确认是否替换还是使用老的，以我的orangepi3b为例，做出如下修改即可使用默认老的文件。

```
diff --git a/external/config/sources/families/rockchip-rk356x.conf b/external/config/sources/families/rockchip-rk356x.conf
index 755d0d8..623ddcb 100644
--- a/external/config/sources/families/rockchip-rk356x.conf
+++ b/external/config/sources/families/rockchip-rk356x.conf
@@ -45,7 +45,7 @@ family_tweaks_s() {

        rsync -a --chown=root:root "${EXTER}"/packages/bsp/rk356x/* ${SDCARD}/

-       chroot $SDCARD /bin/bash -c "apt-get -y -qq install dnsmasq v4l-utils cheese swig python3-dev python3-setuptools bluez libncurses-dev" >> "${DEST}"/${LOG_SUBPATH}/install.log 2>&1
+       chroot $SDCARD /bin/bash -c "apt-get -y -qq install dnsmasq v4l-utils cheese swig python3-dev python3-setuptools bluez libncurses-dev -o Dpkg::Options::="--force-confold"" >> "${DEST}"/${LOG_SUBPATH}/install.log 2>&1

        [[ ${RELEASE} == jammy ]] && rm ${SDCARD}/etc/profile.d/orangepi-ssh-title.sh # for adbd
        chroot "${SDCARD}" /bin/bash -c "systemctl --no-reload enable usbdevice.service >/dev/null 2>&1"
```

baiywt, 2025-02-17:
已修复https://github.com/orangepi-xunlong/orangepi-build/commit/23158a3ecc844ebc60e675d95393b016fdb91619

closed 2025-03-17 by orangepi-xunlong